microsoft/AirSim #3423: Bug with Image Segmentation Colors

tiusty, 2021-02-25:
## Bug report 

- AirSim Version/#commit: 288c90ca78ce09ed58de8508622cb97071c8f217
- UE/Unity version: 2.45
- autopilot version: N/A
- OS Version: OSX Catalina

### What's the issue you encountered?
The Image segmentation colors changed when upgrading to latest version of Airsim. Unfortunately I am not sure what commit I was using before but the priori version I had pulled master around September. I still have a world with the old Airsim plugin so if there is a way to extract the commit for Airsim from that I can provide the old version of Airsim that I was using.

Environment: Mountain Landscape

With the September Airsim Code:
[image]

With the Latest Airsim Code:
[image]
 
As suggested in this page: https://microsoft.github.io/AirSim/image_apis/#segmentation I set the object ID for the bright pink lake in the original segmentation to object id 37.
According to the RGB values to Object id file (https://microsoft.github.io/AirSim/seg_rgbs.txt) this maps to RGB 254,12,229

I am use that RGB value to extract out the Lake and everything works fine with the old version of Airsim.

Now when I upgraded Airsim, using the same exact code, i.e it sets the same static mesh to object ID 37, it now fails to extract the lake because the RGB values have changed. The RGB value for the lake is now 253,0,200.

Therefore, it seems that the color mapping file is no longer accurate and the colors changed. Was this intentional? I do not see any updates on the segmentation color mapping file. 

### Settings
The settings are the same for both configurations

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1080,
        "Height": 1080,
        "FOV_Degrees": 90
      },
      {
        "ImageType": 5,
        "Width": 1080,
        "Height": 1080,
        "FOV_Degrees": 90
      }
    ]
  }
}

### How can the issue be reproduced?
To reproduce:
1. First call client.simSetSegmentationObjectID(37, "SM_IcePlane_9") so the main lake is the proper object ID 
2. Using Unreal 4.25 with the latest Airsim commit, use client.simGetImages with image type airsim.ImageType.Segmentation to retrieve the image of the Mountain Landscape environment
3. Now load the image in opencv or Pillow load the image to numpy array
4. Inspect the image and go to the location in the image that contains the lake and verify the RGB values do not line up with the color pallet on the airsim website.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2021-02-28:
This is similar to #3103.
I was able to replicate the issue and found that the RGB values shown in the viewport are correct:
[image]
while those in the saved images are not:
[image]

TidalPaladin, 2021-03-01:
I'm having similar issues with the 1.4 Linux binaries. I found that while the saved image RGBs do not match those in [seg_rgbs.txt](https://microsoft.github.io/AirSim/seg_rgbs.txt), they are consistent across the Linux binaries I tested. As a workaround I wrote the following script that checks the class -> RGB mapping, and I'm using the output of this file in place of seg_rgbs.txt.

```python
#!/usr/bin/env python
# -*- coding: utf-8 -*-
"""Helper script to check the class -> color mapping in AirSim.

Usage:
    1. Set segmentation resolution to a low value in settings.json (10x10)
    2. Launch an Unreal scene and run this script
    3. Output `seg_colors.csv` will be written with a class -> RGB mapping
"""
import airsim
import numpy as np
import csv

client = airsim.VehicleClient(timeout_value = 7200)
client.confirmConnection()

requests = airsim.ImageRequest("0", airsim.ImageType.Segmentation, False, False)

colors = {}
for cls_id in range(256):
    # map every asset to cls_id and extract the single RGB value produced
    client.simSetSegmentationObjectID(".*", cls_id, is_name_regex=True)
    response = client.simGetImages([requests])[0]
    img1d = np.frombuffer(response.image_data_uint8, dtype=np.uint8)
    img_rgb = img1d.reshape(response.height, response.width, 3)

    color = tuple(np.unique(img_rgb.reshape(-1, img_rgb.shape[-1]), axis=0)[0])
    print(f"{cls_id}\t{color}")
    colors[cls_id] = color

with open('seg_colors.csv', 'w') as f:
    writer = csv.writer(f, delimiter=' ')
    for k, v in colors.items():
        writer.writerow([k] + list(v))
```

This script gave me the following class id -> RGB mapping
<details><summary>mapping</summary>
<p>

```
0 0 0 0
1 0 38 81
2 132 36 41
3 16 48 25
4 13 192 131
5 11 131 157
6 4 1 21
7 139 110 43
8 2 104 90
9 116 101 61
10 145 2 3
11 220 1 33
12 73 37 226
13 2 144 84
14 89 25 7
15 44 178 14
16 0 214 0
17 0 3 140
18 2 14 48
19 13 9 0
20 15 8 73
21 70 76 194
22 104 53 78
23 0 5 137
24 90 47 155
25 11 8 168
26 160 20 166
27 128 61 129
28 155 16 9
29 86 249 34
30 50 1 51
31 15 58 1
32 58 216 139
33 109 251 28
34 24 245 119
35 226 92 26
36 0 3 145
37 200 0 253
38 231 140 4
39 7 93 182
40 171 253 23
41 56 0 80
42 36 3 27
43 26 200 159
44 7 17 52
45 17 9 31
46 218 56 53
47 39 81 186
48 4 80 23
49 3 15 106
50 52 55 4
51 3 1 10
52 47 10 60
53 43 51 178
54 147 0 47
55 27 57 192
56 1 26 235
57 224 83 8
58 83 30 150
59 119 72 122
60 6 206 29
61 59 231 59
62 3 132 116
63 134 186 0
64 35 226 31
65 180 99 26
66 177 9 13
67 57 13 75
68 154 33 152
69 17 19 175
70 243 1 210
71 2 157 39
72 1 137 94
73 214 2 86
74 24 42 87
75 39 2 233
76 4 1 14
77 27 169 118
78 5 115 102
79 75 126 8
80 40 147 237
81 108 12 2
82 222 2 32
83 71 139 132
84 14 4 249
85 76 18 224
86 67 3 216
87 23 202 47
88 38 4 2
89 253 31 19
90 3 90 42
91 228 8 17
92 251 194 13
93 18 35 6
94 175 210 0
95 33 224 1
96 16 1 27
97 136 255 134
98 94 5 251
99 47 109 2
100 204 228 123
101 30 208 101
102 72 173 228
103 1 63 16
104 33 42 89
105 1 27 171
106 74 66 98
107 118 39 33
108 53 46 9
109 3 112 0
110 31 12 5
111 81 65 125
112 32 159 24
113 196 73 16
114 126 4 214
115 173 0 173
116 13 166 1
117 60 7 42
118 12 5 7
119 54 34 180
120 63 222 10
121 166 59 196
122 149 16 72
123 120 106 177
124 42 5 243
125 14 52 160
126 241 125 190
127 6 75 15
128 1 23 220
129 190 44 38
130 2 171 10
131 6 74 115
132 105 128 7
133 7 22 184
134 178 97 83
135 129 177 12
136 49 116 144
137 101 71 149
138 1 0 58
139 42 129 55
140 41 196 93
141 112 86 6
142 68 23 13
143 93 136 46
144 19 3 1
145 102 6 147
146 0 21 17
147 168 4 11
148 192 4 16
149 5 190 29
150 182 102 22
151 106 10 88
152 216 26 1
153 22 29 3
154 239 149 4
155 162 16 12
156 37 212 56
157 5 43 204
158 29 17 4
159 82 198 218
160 9 168 11
161 96 28 1
162 0 108 67
163 255 98 66
164 26 27 45
165 125 0 120
166 11 2 96
167 137 218 71
168 5 87 0
169 210 39 18
170 48 0 77
171 235 0 68
172 144 62 202
173 208 11 0
174 20 243 206
175 10 41 70
176 84 15 126
177 12 123 0
178 194 188 64
179 186 120 49
180 0 233 97
181 66 0 109
182 164 11 222
183 0 6 239
184 8 21 22
185 98 239 188
186 77 1 24
187 45 113 40
188 2 22 142
189 51 160 212
190 6 18 0
191 8 2 110
192 245 88 82
193 159 40 76
194 1 204 99
195 150 237 162
196 80 155 20
197 169 184 113
198 4 0 2
199 8 45 231
200 131 14 37
201 65 145 2
202 99 24 104
203 10 150 62
204 23 6 0
205 110 54 36
206 142 241 108
207 64 105 105
208 21 68 198
209 123 154 0
210 237 11 2
211 19 32 208
212 36 13 54
213 87 24 3
214 249 84 65
215 62 0 5
216 115 31 1
217 188 77 19
218 122 119 57
219 4 50 39
220 29 64 11
221 28 29 164
222 1 7 50
223 92 142 63
224 0 1 128
225 78 180 5
226 61 70 154
227 16 19 241
228 113 13 18
229 18 235 5
230 0 82 1
231 1 60 44
232 212 49 0
233 202 36 112
234 0 47 1
235 31 0 35
236 9 57 8
237 2 134 3
238 97 96 2
239 21 89 0
240 57 6 36
241 34 175 6
242 184 33 255
243 140 78 3
244 25 182 169
245 198 162 15
246 22 122 30
247 206 220 21
248 55 94 245
249 233 0 136
250 88 2 4
251 152 67 200
252 9 10 57
253 0 164 74
254 46 152 92
255 255 255 255
```
</p>

ahmed-elsaharti, 2021-03-01:
@TidalPaladin that's a nifty workaround 👍. Your script's output would be a good replacement for `seg_rgbs.txt` until the reason for the difference in values is tracked down.

rajat2004, 2021-03-01:
There's this old PR https://github.com/microsoft/AirSim/pull/2845 which fixed some bias in the SurfaceNormals and Segmentation images. I wasn't able to check it at the time, but maybe the mapping was correct earlier? But the comments seem to indicate that wasn't the case, and there were issues before and after

@TidalPaladin That seems like a nice way to generate the mapping! 

One place which might a good place to look at is the Segmentation assets, there seem to be 2 of them - [SegmentationMaterial.uasset](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Content/HUDAssets/SegmentationMaterial.uasset) and [seg_color_palette.uasset](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Content/HUDAssets/seg_color_palette.uasset). Could provide more info as to actual values being used if that's stored somewhere

ahmed-elsaharti, 2021-03-01:
The 37th color in the `seg_color_palette.uasset` asset is also consistent with the expected RGB values similar to what's seen on the UE viewport.
[image]

Could there be some sort of compression on the client-side that is causing this?

rajat2004, 2021-03-01:
I just checked the new values mentioned in https://github.com/microsoft/AirSim/pull/2845#discussion_r459867489 and compared it against the current [segs_rbg.txt](https://microsoft.github.io/AirSim/seg_rgbs.txt), the new ones appear in the file while the older ones don't (for e.g. search for 103 in the file). So seems like the PR fixed the values to align with the ones mentioned in segs_rgb,txt

So the values as mentioned in the PR seem to be correct, as well as the ones in viewport, they're both in the file.

@ahmed-elsaharti Are you first saving the images and then opening them in some viewer and comparing the values? The viewer might be causing some difference in that case, or unlikely case if the image is saved as PNG there's some change, but since its lossless compression that shouldn't happen. The latest master saves non-compressed images as PPM so that could be tested as well

ahmed-elsaharti, 2021-03-01:
@rajat2004 the viewport and `seg_color_palette.uasset` values up there (ie: the ones matching those in segs_rbg.txt) are taken from screenshots. While the values in the second image [here](https://github.com/microsoft/AirSim/issues/3423#issuecomment-787485623) are taken from a saved png.

Last thing I found so far was that `response.image_data_uint8` in [UnrealImageCapture.cpp](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/UnrealImageCapture.cpp) also contains the 'wrong' values. This is leading me to believe that this is not due to something on the python client end but more on AirSim's end.

rajat2004, 2021-03-01:
Ah, that's bad. So at the time of PR, the values are correct. And now they're not. Some testing of older commits seems to be best way forward, https://github.com/microsoft/AirSim/pull/3053 is the only PR which modified the assets. Previous to that 4.24 should work as well, to check if UE version made some difference

rajat2004, 2021-03-01:
Will update this comment with my testing

<details>
<summary>Settings</summary>

```json
{
    "SettingsVersion": 1.2,
    "SimMode": "ComputerVision",
    "Vehicles": {
        "MyVehicle": {
            "VehicleType": "ComputerVision",
            "Cameras": {
                "my_camera": {
                    "CaptureSettings": [
                        {
                            "ImageType": 5,
                            "FOV_Degrees": 90,
                            "Width": 10,
                            "Height": 10
                        }
                    ],
                    "X": 0.0,
                    "Y": 0.0,
                    "Z": 0.0,
                    "Pitch": 0.0,
                    "Roll": 0.0,
                    "Yaw": 0.0
                }
            }
        }
    }
}
```

</details>

Script as commented - https://github.com/microsoft/AirSim/issues/3423#issuecomment-788077713 with camera name changed

1. UE 4.24, Commit https://github.com/microsoft/AirSim/commit/d9289cf6a09e982a01cfd6867bd40417588a8d97
[seg_colors.txt](https://github.com/microsoft/AirSim/files/6064156/seg_colors.txt)
Correct results, matches with `segs_rgb.txt`

closed 2021-08-09 by jonyMarino