microsoft/AirSim #3424: How to change AirSim standart drone mesh to my drone or standart car to my car

Supaple-x, 2021-02-28:
## Drone and Car mesh cange
### I want to change standart drone or car mesh to my visualisation of the drone and the car

### I try to achive the right visualisation of my drone when I use AirSim

#### My model has correctly rotating propellers and the model looking as my real drone.

### I have tried to find the actor of standart AirSimDrone and change it in ue. But I didn't find.
Do someona have any tutorial how to perform this customization?

Thanks

ahmed-elsaharti, 2021-02-28:
Hi @Supaple-x , to change the drone/car mesh in AirSim you'd have to modify the AirSim uasset for the vehicle or create a similar one. To do so, have a look at #3190, [this guide](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#pawnpaths) and this video:
https://drive.google.com/file/d/1-iOfJ8zO3YA4Mzc0ZkKwTM3L4--4RFtt/view?usp=sharing

Supaple-x, 2021-02-28:
Hi @ahmed-elsaharti ,
Thank you! This tutorial compleatly meets my targets.
Very detailed and accurate.
All the best!

closed 2021-02-28 by Supaple-x