microsoft/AirSim #3428: settings.json is searched for in an unexpected location. A fix?

nabeelsherazi, 2021-03-02:
## What feature are you suggesting?
### Overview:
Users expect that settings.json should be placed in the root directory of the built environment (e.g. `.../Blocks/`). However, Airsim actually searches the directory where the deep binary resides (`.../Blocks/Blocks/Binaries/Win64/`) which I think is unexpected behavior. See below for more explanation.

### Smaller Details: 
One of the methods of providing a settings.json to Airsim is to place the file in the same directory as the executable of the environment. This is particularly useful if you have multiple environments on your machine, and want to keep seperate settings for each without having to constantly change the default settings.json in `C:/Users/Documents/Airsim/`.

Since users are expected to start the environment using the top level binary, it might be expected that a settings.json should also be provided adjacent to it, in the root folder. However, Airsim will currently not be able to find this settings.json. The folder *actually* searched is that of the deep binary `.../Blocks/Blocks/Binaries/Win64/` (and etc. for other environments). This is unexpected, and I think should be remedied.

I've tested this behavior on three environments from pre-built binaries (Blocks, City, ZhangJiaJie) @ release 1.4.0, and found this behavior in all of them.

The relevant code I believe lives here:
https://github.com/microsoft/AirSim/blob/4a91e0c3eb988d3da3fab7c7d6a85c9ee3876252/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L365-L372

particularly the implemenation of `getExecutableFullPath`, which in turn uses `getExecutableFolder`. I think the core of the issue is here"
https://github.com/microsoft/AirSim/blob/4a91e0c3eb988d3da3fab7c7d6a85c9ee3876252/AirLib/src/common/common_utils/FileSystem.cpp#L98-L116

where `GetModuleFileNameW` will get the filename of the executable which is actually running (the deep binary) rather than the executable which spawned it. This is where I'm out of my element in how to solve the problem, as I'm super unfamiliar with the workings of win32api and libloaderapi.h (which is where `GetModuleFileNameW` comes from.

Also of note is that I have not tested this on Mac or Linux, and am unsure if the issue is present on those platforms too.

### Nature of Request:  

- Change
  - 

## Why would this feature be useful?
The docs would come more in line with user expectations for where settings.json should be placed. I've seen [at least one issue](https://github.com/microsoft/AirSim/discussions/3354) where folks have been confused by how to modify settings.json. I think if a method could be found to have it search the originating executable rather the one the one all the way down there, it could  be much more convienent to modify settings.

Thank you for reading!

jonyMarino, 2021-03-02:
Thanks, @nabeelsherazi! We will deal with this as a bug.

nabeelsherazi, 2021-03-02:
Thanks! I was reticent to bill it as a bug because the docs statement "a file placed in the same directory as the binary" *is* technically true. But if folks think this was not what was intended by that, then a bug it is I suppose 😄

rajat2004, 2021-03-03:
@nabeelsherazi Could you perhaps test out https://github.com/microsoft/AirSim/pull/3436? I haven't yet built and tested it on a binary yet, but a sanity check on Linux for the method seems to work

PR #3436 (2021-06-04): Check for settings.json in current directory as well

Fixes: #3428    

## About

Adds a check for `settings.json` in the directory from where the binary was launched
For e.g. in Linux, if invoking `Blocks.sh` from inside the `LinuxNoEditor` folder, the path searched would be `.../LinuxNoEditor/settings.json`. If started from a level above using `./LinuxNoEditor/Blocks.sh`, it would be `.../settings.json`.
 
Currently, it's an additional check rather than replacing the executable location since that'll be a behavioural change, can be removed if required.

Also, this again brings up the usefuln [... cut]
closed 2021-06-04 by zimmy87