microsoft/AirSim #3430: Setting gimbal angle using the ROS wrapper resets camera position

bedaberner, 2021-03-02:
## Bug report 

- AirSim Version/#commit: Master
- UE/Unity version:4.25
- autopilot version:PX4 1.10.1
- OS Version:Linux 18.04

### What's the issue you encountered?
When setting the gimbal angles using /airsim_node/gimbal_angle_euler_cmd the position of the camera is reset to the drone center. I tracked the error to this line
https://github.com/microsoft/AirSim/blob/4a91e0c3eb988d3da3fab7c7d6a85c9ee3876252/ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp#L1193
where the camera position is set to 0 instead of the coordinates defined in the settings.json,
### Settings
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "Copter": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            },
            "Cameras": {
				"Camera": {
					"CaptureSettings": [
						{
							"ImageType": 0,
							"Width": 640,
							"Height": 480,
							"FOV_Degrees": 90
						},
						{
							"ImageType": 5,
							"Width": 640,
							"Height": 480,
							"FOV_Degrees": 90
						}
					],
					"Gimbal": {
						"Stabilization": 1,
						"Pitch": 1, "Roll": 1, "Yaw": 1
					},
					"X": 0,
					"Y": 0,
					"Z": 2,
					"Pitch": 0,
					"Roll": 0,
					"Yaw": 0
				}
			}
        }
    }
}

### How can the issue be reproduced?

1. Define a camera with coordinates other than 0 0 0 and a gimbal
2. set a new gimbal orientation using the ros wrapper

bedaberner, 2021-03-11:
If anybody else has this problem, a hacky solution is to hardcode the desired gimbal location at the mentioned line to the same value as the settings.json

jonyMarino, 2021-03-22:
Hi @bedaberner! and Welcome! PRs are welcome too.