microsoft/AirSim #3437: about the function of "car_controls"

dana4056, 2021-03-03:
## Question
### What's your question?
Isn't "car_controls.throttle" speeding up the car so it can move?

I want to drive a car only through the API without using a keyboard. 

Therefore, I gave "RemoteControlID" a value of 1 as shown in the json file below (because the document says that if I give more than 0 value, remote control connected to the system is possible), and gave "1"  to "car_controls.throttle" 

But when I run unity, I can control the keyboard, and when I run the python file, the car doesn't move according to the code.

Is there anything I'm missing?

### Include context on what you are trying to achieve
hello_car.py
```
import airsim
import cv2
import numpy as np
import os
import setup_path 
import time

# connect to the AirSim simulator 
client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = airsim.CarControls()

car_controls.throttle = 1.0
```

settings.json
```
{
  "SimMode": "Car",
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": 1
      }
    }
  }
}
```

#### Context details
windows 10 
unity 2019.3.12

### Include details of what you already did to find answers

dana4056, 2021-03-03:
I forgot.
python 3.8.5
airsim v1.4.0

dana4056, 2021-03-04:
> @dana4056 you missed the setting call : `client.setCarControls(car_controls)`
> Also, `import setup_path` needs to be before `import airsim` (not related to this particular issue you're facing though)

Thank you for your advice.
I did what you said.
The code is as below. But the same thing, the car doesn't move.

```
import setup_path
import airsim
import cv2
import numpy as np
import os
import time

# connect to the AirSim simulator 
client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = airsim.CarControls()

for idx in range(3):
    # get state of the car
    car_state = client.getCarState()
    print("Speed %d, Gear %d" % (car_state.speed, car_state.gear))

    # go forward
    car_controls.throttle = 0.5
    car_controls.steering = 0
    client.setCarControls(car_controls)
    print("Go Forward")
    time.sleep(3)   # let car drive a bit

    # Go forward + steer right
    car_controls.throttle = 0.5
    car_controls.steering = 1
    client.setCarControls(car_controls)
    print("Go Forward, steer right")
    time.sleep(3)   # let car drive a bit

    # go reverse
    car_controls.throttle = -0.5
    car_controls.is_manual_gear = True;
    car_controls.manual_gear = -1
    car_controls.steering = 0
    client.setCarControls(car_controls)
    print("Go reverse, steer right")
    time.sleep(3)   # let car drive a bit
    car_controls.is_manual_gear = False; # change back gear to auto
    car_controls.manual_gear = 0

    # apply brakes
    car_controls.brake = 1
    client.setCarControls(car_controls)
    print("Apply brakes")
    time.sleep(3)   # let car drive a bit
    car_controls.brake = 0 #remove brake

#restore to original state
client.reset()

client.enableApiControl(False)
```

ahmed-elsaharti, 2021-03-04:
Woops, i just realized that you're using Unity rather than Unreal Engine.
This is probably similar to #2679 which is addressed in #2937

dana4056, 2021-03-04:
> Woops, i just realized that you're using Unity rather than Unreal Engine.
> This is probably similar to #2679 which is addressed in #2937

Thank you so much. I followed the link you attached and read it carefully, and the solution to the content is related to the C++ API. Of course, there was a python API, but it didn't exactly match my problem. Maybe, I can't find it. :'(

jonyMarino, 2021-03-22:
Hi @dana4056! Let's continue in #2679.

closed 2021-03-22 by jonyMarino