microsoft/AirSim #3439: How to use the pre-compiled binaries for setting up the city environment for win10

AlexanderZ3, 2021-03-04:
## Question
### What's your question?

Main Question: How to use the pre-compiled binaries AirSimNH for setting up the city environment for Win10?

#### Setting environments 
 
Win 10,  AirSim1.4 (Latest), 
UE2.5 and 2.4 (I installed both engines, 2.5 can run block successfully and 2.4 for landscape mountain. But neither can work for the AirSimNH. )

### Include context on what you are trying to achieve
 I want to fly the multirotor in the urban neighborhood block environment, but when I unzip the downloaded AirSimNH files, it fails to be compiled. My operation procedures are shown in the below figures:
1. unzip:  AirSimNH >> LinuxNoEditor >> AirSimNH >> AirSimNH.uproject 
[image]
2. Double click the AirSimNH.uproject (The default is UE2.5, I also tried UE2.4, Still not work)
[image]

### Include details of what you already did to find answers
I have run the block environment in UE2.5 successfully, and the landscape mountain in UE2.4. But there are some strange points, shown in the below figures, I am not sure if they influence the main question. If not, could you give me some advice for these two problems?
1. some warnings when building every project: "Shell32.lib", "dinput8.lib" and "dxguid.lib" are not resolvable.
[image]
2. For my interaction page, I don't find the three views shown in the bottom position in the demo video. How to obtain these views?
[image]
[My interaction page]
[image]

Thanks a lot!

AlexanderZ3, 2021-03-04:
Btw, I want to add two basic questions:
1. do I need to use my locally built Airsim Plugins file to replace the original plugins file in the pre-built binaries file?
2. does the LinuxNoEditor mean that t can only operate on the Windows platform? cause I notice some environments have WindowsNoEditor (such as the AbandonedPark)?

Maybe these are just some beginning-level questions, hope you can give me some clues. Thanks for your help, sincerely!

bedaberner, 2021-03-04:
to get the three views press 1, 2 and 3 while the simulation is running ( F1 displays help btw)
for your other questions:
1. the pre-built ones should work
2. LinuxNoEditor means it was built for Linux, WindowsNoEditor for Windows. If you download the linux versions it should always be called LinuxNoEditor

ahmed-elsaharti, 2021-03-04:
To add to @bedaberner's answer and to tackle your main issue:
Based on your screenshots it looks like you downloaded Linux binaries rather than Windows ones and are trying to run those on Windows which won't work.
<details>
<summary>As seen here, there are two releases per version of AirSim</summary>
[image]
</details>
And AFAIK the Windows 1.4.0 release has 2 environments that are not present in the Linux 1.4.0 release

Moreover, released binaries are pre-compiled. ie: you don't need to compile them nor edit them with the Unreal Editor (AFAIK there's no way to edit them actually).
All you need to do is download the correct binary for Windows (in your case) and doubleclick the executable (.exe) file to launch:
[image]

AlexanderZ3, 2021-03-05:
@bedaberner @ahmed-elsaharti Thanks, your answers help me a lot! Problems are solved and the Neighbour environment runs smoothly.
During the running, I noticed two more questions,
1. Besides run the .exe file, can I add my code and operate the drone by following my program? can you recommend any tutorial example?
2. During the flying, the multirotor can detect the collision with some big objects, like a house roof or thick trunk. But I found that for some thin objects, like power lines or bush leaves, the drone just passes through them without collision remindings. I want to know can we define how large or thick one objective can be viewed as a collision?  Are there some adjustable parameters for this purpose? 

Thanks for your help!

AlexanderZ3, 2021-03-05:
Hi @ahmed-elsaharti @bedaberner   Luckily, I think I have figured out question 1, by using the hello_drone.py example.  Question 2 is still unknown ~~

BTW, one more question, when I call the hello_drone.py, everything runs smoothly, but I found the interaction shows"API call is not received, entering hovering for safety", as shown in the below figure. Actually, I think the API call is received, since the drone flies as programmed, and it does not enter hover mode. Although this reminding seems not to influence anything, I am just curious if this is normal. Maybe you also met similar problems before?
[image]

ahmed-elsaharti, 2021-03-08:
Hi @AlexanderZ3 , to answer this:
> During the flying, the multirotor can detect the collision with some big objects, like a house roof or thick trunk. But I found that for some thin objects, like power lines or bush leaves, the drone [...]

I think that is a collision setting from within the Unreal Engine environment itself

Regarding your last question, I don't have a very convincing answer but I think maybe this happens between the takeoff call and the movement call and thats why you dont notice it hovering?

AlexanderZ3, 2021-03-10:
Hi @ahmed-elsaharti ， Thanks!

Actually, "API call is not received, entering hovering for safety" shows all the time. I will keep trying and find more clues. Thanks again.