microsoft/AirSim #3442: Lost python files when call python API

AlexanderZ3, 2021-03-04:
## Question
### What's your question? 
Could anyone tell me where to get the lost files: character_control.py, DQNdrone.py? 
 
When I call the python API in VS2019, it pops up below errors:
[image]
Furthermore, I found the "DQNdrone.py" is also lost in the multirotor file. I follow the tutorial  https://airsim-fork.readthedocs.io/en/docs/apis.html#python-quickstart  and the airsim is built successfully. So could anyone know why this problem happen and how to solve it? Thanks a lot!
[image]

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

ahmed-elsaharti, 2021-03-04:
Hi @AlexanderZ3 , try following the [updated docs](https://microsoft.github.io/AirSim/apis/#python-quickstart).
The easiest way to run the demo python scripts is (as mentioned in the docs) to run `python scriptName` in your terminal (with python installed and the airsim and msgpack-rpc-python packages installed ofcourse) like:
`python hello_car.py` for example.

Nonetheless, the inclusion of outdated scripts that no longer exist will probably need to be corrected for in [PythonClient.pyproj](https://github.com/microsoft/AirSim/blob/master/PythonClient/PythonClient.pyproj)

AlexanderZ3, 2021-03-05:
Hi @ahmed-elsaharti ， thanks for your reply! Problems are solved! This comment can be closed.

closed 2021-03-05 by AlexanderZ3