microsoft/AirSim #3444: propeller_area in createDragVertices use diameter instead of radius, xsection use full propeller instead of half

clementicus, 2021-03-05:
Hi,

I think the propeller area formula in [AirLib/include/vehicles/multirotor/MultiRotorPhysicsBody.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorPhysicsBody.hpp#L181) is wrong.

```
real_T propeller_area = M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_diameter;
```

Should rather be params.rotor_params.propeller_diameter **/ 2**?
```
real_T propeller_area = M_PIf * params.rotor_params.propeller_diameter / 2 * params.rotor_params.propeller_diameter / 2;
```

In the same way, i'm not sure I understand the next line : 
```
real_T propeller_xsection = M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_height;
```
Should not we use only half of the xsection of the propeller, the part exposed ?
```
real_T propeller_xsection = M_PIf * params.rotor_params.propeller_diameter / 2 * params.rotor_params.propeller_height;
```

ahmed-elsaharti, 2021-03-06:
Hi @clementicus , for the first equation, I think `propeller_area` is accounting for the total area of all rotors.
ie:
[image]
I'm guessing this based on `drag_factor_unit` where the value of `propeller_area` is used as is and is not multiplied by the number of rotors which probably means that its already accounted for.
as for the second equation, it seems to me that since `propeller_xsection` is added to `left_right_area` in `drag_factor_unit`, based on the name of `left_right_area` I'm assuming that this is for both sides, ie: the area exposed on the right (half props' xsection) and the area exposed on the left (another half, of two other propellers, but numerically the same values).

I'm just guessing this based on what I'm seeing in [MultiRotorPhysicsBody.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorPhysicsBody.hpp) I could be completely wrong though.

The one actual problem I see with this though is that it (if my explanation up there is true) assumes that the rotor count is always four. Which is not the case if we're simulating a hexacopter.

clementicus, 2021-03-07:
Hi, @ahmed-elsaharti, thanks for your answer !

If I'm correct, in AirSim paper, the drag equation is described by :

[image]

where A is the vehicle cross-section.

The drag_factor_unit is multiplied in [FastPhysicsEngine.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/physics/FastPhysicsEngine.hpp) by [image] , which means that ```drag_factor_unit``` should be:

[image]

so 

[image] corresponds to the term ``` * params.linear_drag_coefficient / 2 ```

This means that all other terms should be the area [image].

### Front back and Left Right
For front_back (and similarly for left_right): [image] should be equal to:
```front_back_area + rotors_.size() * propeller_xsection```

My issue/question is about the ```propeller_xsection```: it is the cross section exposed of each rotor, which is the half of the cross-section.
Knowing that ``` real_T propeller_xsection = M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_height``` ,
We should have only he half. So for front_back (and similarly for left_right): [image] should be equal to:
```front_back_area + rotors_.size() * propeller_xsection/2```

### Top bottom
For top bottom area, the [image] term is 
```top_bottom_area + rotors_.size() * propeller_area```

This is ok, but the question is about the proppeler_area: the area of the propeller is [image], and not [image]

```real_T propeller_area``` should be then: 
```M_PIf * params.rotor_params.propeller_diameter /2 * params.rotor_params.propeller_diameter /2```

what do you think?

ahmed-elsaharti, 2021-03-08:
@clementicus , you're right! I tracked this down to af655fe482f8e03a15ac562a72470f0b6fe394a1 where the name of the variable was changed in [MultiRotor.hpp](https://github.com/microsoft/AirSim/blob/af655fe482f8e03a15ac562a72470f0b6fe394a1/AirLib/include/vehicles/MultiRotor.hpp) and [RotorParams.hpp](https://github.com/microsoft/AirSim/blob/af655fe482f8e03a15ac562a72470f0b6fe394a1/AirLib/include/vehicles/RotorParams.hpp) (ironically, when it was labeled 'radius' it still contained the value of the **diameter** of the DJI phantom 2 rotor, 0.2m). Easiest fix, as you mentioned, would be to change the equations.

Also, now that I looked at the drag force formula, I have an entirely different issue with `propeller_xsection`:
From what I understand, the drag force equation expects A (in the case of front-back and left-right) to be the cross-sectional area rather than the surface area.
`M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_height` is the surface area of the rotor (which we're assuming is a disk at this point). while the cross-sectional area of the disk should be more like `params.rotor_params.propeller_diameter * params.rotor_params.propeller_height` and that would get multiplied by rotors_.size() and divided by two (since as you mentioned we see half the total number of rotors when looking from one side).

[image]
or am I missing something?

clementicus, 2021-03-08:
Hi @ahmed-elsaharti . It is a good question.

Looking at litterature I found that we can use both method as long as the coeficient is defined accordingly ([NASA edu](https://www.grc.nasa.gov/www/k-12/rocket/drageq.html)) ...

The "cross section area" option is when the drag coefficient is a "frontal drag coefficient" and in this case the surface is projected in a normal plane to the direction of the fluid far from the object, which in this case we should use what you said: 
```params.rotor_params.propeller_diameter * params.rotor_params.propeller_height```

Therefore the question is which one of the two options is used in Airsim.
I see in [MultiRotorParams.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp) a comment and reference to this page: http://physics.stackexchange.com/q/304742/14061 where we see the equation used for Friction Torque: 
[image]
and the comment:  "Also note that [image] might be only half because only half the surface area is experiencing fluid in surface normal".
What I understand is that we use the surface (and not the cross section), i.e. ```M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_height```and also we find a confirmation for the half of the section to be used (related to my first issue).

Even if this comment talks about `angular_drag_coefficient` they consider that it is the same as `linear_drag_coefficient` so maybe we can conclude that we need to consider the surface area and not the cross section like you proposed.

Do you agree?

ahmed-elsaharti, 2021-03-09:
It would make sense that the answer on http://physics.stackexchange.com/q/304742/14061 matches the method used in MultiRotorPhysicsBody.hpp since Shital was the one who wrote both and is using the surface area in both for the calculation of drag force.

However, the [drag coefficient being used here](https://github.com/microsoft/AirSim/blob/b5d2f257ba024e30ecc6e19e50e694e0d31a5a80/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L39-L42) comes from [this source](http://klsin.bpmsg.com/how-fast-can-a-quadcopter-fly/) that states that they're using projections for simplifications.
Moreover, every other source I've found considers the cross-sectional area facing the flow (refered to as the characteristic frontal area in some references):
https://www.engineeringtoolbox.com/drag-coefficient-d_627.html:
"[image]"
(the examples in their table also support this)

https://en.wikipedia.org/wiki/Drag_coefficient:
"[image]"

https://www.lmnoeng.com/Drag/index.php (the cone example is a strong example showing this)

This would be consistent with the comments in AirSim since (from my understanding) for simplicity the 'front-facing/form drag' is considered rather than the 'skin-drag' effect (if this was considered, the side faces would have been accounted for in the frontal area too, since they technically are a part of the surface area the air would come in contact with while flying forward similar to how a rounded disk (the prop) would have non-flat parts that we're accounting for by using the surface area equation).

It's actually very confusing that the variable's name is `xsectional_area` yet it contains the surface area 😂

clementicus, 2021-03-16:
Hi @ahmed-elsaharti ,
I think you are right when you say :
> the 'front-facing/form drag' is considered rather than the 'skin-drag' effect (if this was considered, the side faces would have been accounted for in the frontal area too, since they technically are a part of the surface area the air would come in contact with while flying forward...

but maybe they don't consider the side faces because they probably generate a very little drag comparing to the frontal one and propeller. 

In the PR for the first issue I will propose to add a comment saying that the `linear_drag_coefficient` must be calculated based on the surface area 
is we use `M_PIf * params.rotor_params.propeller_diameter * params.rotor_params.propeller_height`