microsoft/AirSim #3445: function/library for the drone to follow waypoints using my own localization algorithm.

charyyev, 2021-03-06:
I implemented localization algorithm that outputs the pose of the drone given an image. I want to demonstrate that my algorithm is accurate enough to use for navigation on airsim. I want the drone to follow waypoints using my own localization algorithm. I am not good at control and path planning, can you suggest me any library/function that I can use on airsim to follow waypoints while updating drone pose with my own algorithm?

Thanks in advance!

ahmed-elsaharti, 2021-03-08:
Hi @charyyev, I think the [moveOnPathAsync](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L945) API would be a good place to start. [Here's](https://github.com/microsoft/AirSim/blob/f423d9f5a4a0d0f4fb9a2e4ddb75251040e88f81/PythonClient/multirotor/survey.py#L69-L70) an example of how to use it

charyyev, 2021-03-09:
Hi @ahmed-elsaharti, I cannot input the result of my localization algorithm into moveOnPathAsync API. I want something like offboard mode of pixhawk controller. I provide pose of the drone and the path, and  want the drone to follow it.

ahmed-elsaharti, 2021-03-09:
Hi @charyyev , I might havr misunderstood, but isnt the output of your algorithm a set of x, y and z points that you want the simulated drone to follow?