microsoft/AirSim #3448: Issue with RotorActuator parameter when emitting the signal during update of rendering of drone

xxEoD2242, 2021-03-06:
## Bug report 

- AirSim Version/#commit: 1.4
- UE/Unity version: 4.25
- autopilot version:
- OS Version: Ubuntu 20.04

### What's the issue you encountered?
When running a fresh pull, i.e last night (March 5th), I compiled the binaries of LandscapeMountains from an older project and then ran it. Every time a Segmentation fault and the following outtput.

[image]

### Settings

### How can the issue be reproduced?

1. Compile a project using the latest pull of AirSim
2. Run the latest pull of AirSim

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

xxEoD2242, 2021-03-06:
I am currently getting around this by commenting out the offending `.emit()` call and will continue to look into why the vector assignments have mismatched memory access.

xxEoD2242, 2021-03-06:
Well, I'm not sure if this matters, but I wasn't checkout out of the Linux release tag. So, I'm doing that now. Will update on what happens when I compile with this.

rajat2004, 2021-03-07:
Yup, tested this and I'm also seeing this error. I've added a lot of debug statements and I _think_ that I've figured out the problems, but would be great if you could test it as well, the branch is here - https://github.com/rajat2004/AirSim/tree/fix-signal-delete-crash

@ahmed-elsaharti This seems to have occurred after #3402, not sure why this didn't come during testing though. Could you perhaps test it as well?

The debug statements are there since need to do some more testing, but the main change is in the `Signal::disconnect` method

xxEoD2242, 2021-03-07:
@rajat2004 I just compiled and tested on Ubuntu20.04. Works like a charm. Thank you!!!

closed 2021-03-07 by xxEoD2242

ahmed-elsaharti, 2021-03-07:
Hi @xxEoD2242 and @rajat2004, I've been trying to repro this in Windows but couldn't. Tried Blocks and an old LascapeMountains project I had but both of them seem to work fine with 9fc8b36. Not entirely sure why but this seems to be occurring only in Linux from what I'm seeing.

rajat2004, 2021-03-07:
@ahmed-elsaharti Thanks for the detailed testing! It's very strange that the problem doesn't seem to be occurring in Windows but happens consistently on Linux. Since the code also isn't something specific to the OS, currently I can only think of some compiler or OS difference in handling the memory access. I'll probably test it out with some small code snippets for my curiosity :)

ahmed-elsaharti, 2021-03-08:
@rajat2004 you're probably right! I've actually seen different compilers handle range-based for loops differently so that's probably the reason this gets handled well-ish (not sure if I'm happy about it doing what it did at this point 😅) on Windows.

PR #3449 (2021-03-08): Fix crash due to access after delete

Fixes: #3448    
Fixes: #3452

## About

This PR fixes a crash due to access after `erase` in the map. #3402 added a method that is only supposed to be called once, and after that removes itself. However, the range-based for loop is not suitable for this and causes a crash after the method returns due to the iterator now essentially being invalid. Also, see https://stackoverflow.com/questions/8234779/how-to-remove-from-a-map-while-iterating-it/8234813

The second commit adds some cleanup, it isn't actually required but makes things clearer. Can be easily d [... cut]