microsoft/AirSim #3452: Blocks environment works for car, crashes on multirotor.

mmuetert, 2021-03-08:
## Bug report 

- AirSim Version/#commit: v1.4.0 - Linux
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

I just installed Unreal Engine and AirSim following the documents. I went to test the install with the Blocks Environment. Following the documentation, I open UE4Editor and selected the Blocks project. When I clicked play, I was prompted to choose car or multirotor. If I chose multirotor, the simulator crashed. If I chose car, the simulator ran, but there was no sound (I am not sure if the was supposed to be any audio). I removed Unreal and Airsim and reinstalled and I get the same error. Am I missing a driver?

### Settings

All default settings.

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

LoginId:9bc192112f8d4e7ebd4bec8c9d56a477-000003e8
EpicAccountId:

Caught signal

libUE4Editor-AirSim.so!common_utils::Signal<std::__1::vector<MultirotorPawnEvents::RotorActuatorInfo, std::__1::allocator<MultirotorPawnEvents::RotorActuatorInfo> > const&>::emit(std::__1::vector<MultirotorPawnEvents::RotorActuatorInfo, std::__1::allocator<MultirotorPawnEvents::RotorActuatorInfo> > const&) [/media/michael/myPassport2/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/common_utils/Signal.hpp:84]
libUE4Editor-AirSim.so!MultirotorPawnSimApi::updateRendering(float) [/media/michael/myPassport2/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Vehicles/Multirotor/MultirotorPawnSimApi.cpp:130]
libUE4Editor-AirSim.so!ASimModeWorldBase::Tick(float) [/media/michael/myPassport2/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimMode/SimModeWorldBase.cpp:147]
libUE4Editor-Engine.so!AActor::TickActor(float, ELevelTick, FActorTickFunction&) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:1007]
libUE4Editor-Engine.so!FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/Actor.cpp:171]
libUE4Editor-Engine.so!FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:283]
libUE4Editor-Engine.so!TGraphTask<FTickFunctionTask>::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:849]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:692]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:586]
libUE4Editor-Core.so!FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr<FGraphEvent>, TInlineAllocator<4u, TSizedDefaultAllocator<32> > > const&, ENamedThreads::Type) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:1477]
libUE4Editor-Engine.so!FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:557]
libUE4Editor-Engine.so!FTickTaskManager::RunTickGroup(ETickingGroup, bool) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1522]
libUE4Editor-Engine.so!UWorld::Tick(ELevelTick, float) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1558]
libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1692]
libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:411]
UE4Editor!FEngineLoop::Tick() [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4844]
UE4Editor!GuardedMain(char16_t const*) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/media/michael/myPassport2/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xe6)
UE4Editor!_start()
[Blocks.log](https://github.com/microsoft/AirSim/files/6098524/Blocks.log)

**What's better than filing an issue? Filing a pull request :).**

rajat2004, 2021-03-08:
This is being fixed by #3449, could you perhaps test it out

mmuetert, 2021-03-08:
I would love to test this, but I am rather new to github. Is there a way to bring your changes into the AirSim I currently have built, or do I need to download your branch and build it per the documentation? Thanks for the help!

rajat2004, 2021-03-08:
It's a pretty small change, just the lines modified in Signal.hpp should be enough, so you could easily just open the file and make the changes yourself. 

Or if you want to try out some git commands :)

```shell
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git fetch origin pull/3449/head:PR/3449; git checkout PR/3449;
./build.sh
```

rptamin, 2021-03-08:
I am getting in the same error, on ubuntu 20.04. I will try out the fix.

mmuetert, 2021-03-08:
The fix seems to have solved my problem. Thank you so much!

PR #3449 (2021-03-08): Fix crash due to access after delete

Fixes: #3448    
Fixes: #3452

## About

This PR fixes a crash due to access after `erase` in the map. #3402 added a method that is only supposed to be called once, and after that removes itself. However, the range-based for loop is not suitable for this and causes a crash after the method returns due to the iterator now essentially being invalid. Also, see https://stackoverflow.com/questions/8234779/how-to-remove-from-a-map-while-iterating-it/8234813

The second commit adds some cleanup, it isn't actually required but makes things clearer. Can be easily d [... cut]
closed 2021-03-08 by mmuetert