microsoft/AirSim #3458: PlayerStart Issues

Efrit16, 2021-03-11:
## Bug report 

- AirSim Version/#commit: v1.4.0 / commit: 40a2bba
- UE/Unity version:  4.26.1
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I started testing the playerStart positions, used the location as shown in the screenshot:
[image]
the simulation ran smoothly but when I pressed stop, this was the screen:
[image]

I tried other Z values and it turns out if I use Z>1088 the car spawns, but when it reaches the ground it disappears, player start position changes : 
[image]
and when I stopped the simulation, this was the screen : 
[image]

It turns out if  z>300 the screen after the simulation stops is buggy . example (z=500):
[image]

Summary: if Z>1088 the car spawns but disappears once it reaches the ground. The screen after the simulation ends is
                 if  1088>Z>300 the car spawns, simulation goes as intended. The screen after the simulation ends is buggy.
                 if Z<300 the car spawns, simulation goes as intended. The screen after the simulation ends is buggy.
### Settings

Default settings.
### How can the issue be reproduced?

1. Launch AirSim from Visual Studio 2019 (F5).
2. Input coordinates in the screenshots.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

denmonz, 2021-03-11:
I've never worked with the car before, but what happens when you set the PlayerStart and Car model at Z=0? Also, could this be an issue with the camera orientation of the PlayerController in UE4?

Efrit16, 2021-03-12:
At Z= 0, simulation runs as it should.

I think it is related to the camera orientation of the PlayerController in UE4, however in some cases the car disappears completely with no possibility of driving so I believe there's a problem somewhere else as well!