microsoft/AirSim #3459: Drone resets position after call to moveToZAsync.

denmonz, 2021-03-11:
## Bug report 

- AirSim Version/#commit: 40a2bbab9e5faebab74c044f240a68b32a16b055
- UE/Unity version: 4.24.3
- autopilot version: N/A
- OS Version: macOS Big Sur v11.2.1
- Xcode Version: 11.7

### What's the issue you encountered?

When running [this code](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/path.py) in my custom environment, the drone is successfully connected to, armed, and takes off, however, when it attempts to run the line:

` print("make sure we are hovering at predefined height of 7 meters...")
   self.client.moveToZAsync(z=self.z, velocity=1, vehicle_name=self.drone_name).join()`

the drone moves up to the position, then falls back to the ground, and the python code hangs on this line (since it hadn't completed).

When I stop the code, I get the following Traceback:
`Traceback (most recent call last):
  File "path.py", line 94, in <module>
    client_flight.fly_the_path()
  File "path.py", line 46, in fly_the_path
    self.client.moveToZAsync(z=self.z, velocity=1, vehicle_name=self.drone_name).join()
  File "/Users/[me]/miniconda3/envs/[env]/lib/python3.7/site-packages/msgpackrpc/future.py", line 22, in join
    self._loop.start()
  File "/Users/[me]/miniconda3/envs/[env]/lib/python3.7/site-packages/msgpackrpc/loop.py", line 22, in start
    self._ioloop.start()
  File "/Users/[me]/miniconda3/envs/[env]/lib/python3.7/site-packages/tornado/ioloop.py", line 863, in start
    event_pairs = self._impl.poll(poll_timeout)
  File "/Users/[me]/miniconda3/envs/[env/lib/python3.7/site-packages/tornado/platform/kqueue.py", line 66, in poll
    kevents = self._kqueue.control(None, 1000, timeout)`

[Here](https://gist.github.com/denmonz/3d1e3201ab4bdec59e0658532bd149dc) is my Conda environment, should it be helpful.

### Settings

[settings.json](https://gist.github.com/denmonz/a69526c6b73e2aa8ec0477d98d40f623)

### How can the issue be reproduced?

1. Launch UE4 game either from the game's .exe, or by hitting "Play" in the UE4 Editor.
2. Once the drone(s) have loaded, run the python script.
3. AirSim Python API will connect (will show the Connected! Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1) messages), the drone will arm, and takeoff, but then reset its position.

### Include full error message in text form
N/A

**What's better than filing an issue? Filing a pull request :).**

denmonz, 2021-03-11:
This issue also occurs on my Windows machine with similar conda environment.

ahmed-elsaharti, 2021-03-11:
Hi @denmonz, I just ran the python script (along with your settings.json file) with 40a2bba checked out on Windows but could not repro the issue.
https://user-images.githubusercontent.com/49626509/110866432-bcd36380-828a-11eb-91f5-394343d018e8.mp4
Could the conda environment be causing this?

denmonz, 2021-03-11:
Hello, @ahmed-elsaharti! Thank you for your response. I've started my conda environments from scratch several times, unfortunately. Is there a .yml file specifically for AirSim that I can base my conda environment off of? I can imagine the issue could be just one out of several packages that I'm just overlooking...

denmonz, 2021-03-15:
I "resolved" my issue by setting the height of the drone (z) to -50 instead of -7. This issue may be because I modified the BP_FlyingPawn.uasset file in Unreal Engine 4 during development. I can imagine somewhere along the lines, I changed "relative" to "world" or the other way around. In any case, I think the "NED" coordinates no longer apply to the drones in my sim.

Unfortunately, I think there's still a bug somewhere. I've again, restarted my conda virtual environment multiple times, and the only way to make this code work was to modify the value from -7 to -50. My other airsim python scripts work as intended as well, which adds to the confusion. A hack to get it to work in the meantime, but I'll update this thread if I determine the problem and find a non-hacky solution.

closed 2021-03-15 by denmonz