microsoft/AirSim #3469: Convert local coordinates to global coordinates (NED)

KabirovIrik, 2021-03-15:
## Question
### What's your question?
Hello everyone.
I get depthmap from image. I convert it to point cloud. But this point cloud is in local coordinates system.
So, how to convert coordinates from this point cloud to global coordinates system?
In this context, global coordinates system - NED (north-east-down) system.

### Include context on what you are trying to achieve

#### Context details

AirSim-v.1.2.2
map - LandScape - old version with powerline towers
local point cloud - https://skfb.ly/6ZHVt 
My code here https://gist.github.com/KabirovIrik/c511e665965b4af7b73172b0ad7386b0 and duplicate .py https://gist.github.com/KabirovIrik/0f6c24e85b0f01270aecbaf7843e0f32

### Include details of what you already did to find answers
I use this code to get depth image
`responses = client.simGetImages(
    [airsim.ImageRequest('0', airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])
response = responses[0]
img1d = np.array(response.image_data_float, dtype=np.float)
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
with open('img2d.npy', 'wb') as f:
    np.save(f, img2d)`

I use code from here https://medium.com/yodayoda/from-depth-map-to-point-cloud-7473721d3f to generate point cloud, but i replace order of coordinates in "return" line.
I try to use `from scipy.spatial.transform import Rotation` and `client.simGetGroundTruthKinematics().orientation`., but it's not work.
`
R = Rotation.from_quat([0, 0, 0.693, -0.72]) # x, y, z, w
R.apply(local_top_tower) + global_pos
`

For example, I compared coordinates of tower's top. I get coordinates of tower's top in NED (I just used ComputerVision mode and moved to this and called client.simGetGroundTruthKinematics()) and get coordinates in local system (I used coordinate of pixel in image). There is code https://gist.github.com/KabirovIrik/c511e665965b4af7b73172b0ad7386b0

Kinematiks to get position and orientation:
`
{   'angular_acceleration': <Vector3r> {   'x_val': 8.868171789799817e-06, 
    'y_val': -1.7474338829767157e-14, 
    'z_val': 2.608952343052806e-07}, 
    'angular_velocity': <Vector3r> {   'x_val': -1.5225546690089686e-07, 
    'y_val': -4.5590446973164944e-08, 
    'z_val': 1.1268485877735657e-07}, 
    'linear_acceleration': <Vector3r> {   'x_val': -4.031422307093635e-08, 
    'y_val': 9.657274091523504e-09, 
    'z_val': 0.0}, 
    'linear_velocity': <Vector3r> {   'x_val': -7.353530673270825e-09, 
    'y_val': -2.7411378056285685e-08, 
    'z_val': 2.564329690812883e-07}, 
    '<b>orientation</b>': <Quaternionr> {   'w_val': 0.8131623268127441, 
    'x_val': -1.0377081416379497e-09, 
    'y_val': 2.0469372863374247e-09, 
    'z_val': -0.5820370316505432}, 
    '<b>position</b>': <Vector3r> {   'x_val': 104.6720962524414, 
    'y_val': 39.37985610961914, 
    'z_val': 0.023845288902521133}} 
`
Tower's top is in screenshots.

[image]
[image]
[image]

KabirovIrik, 2021-03-16:
I made mistake in orientation date

This correct:
'orientation': <Quaternionr> {   
'w_val': -0.7204451560974121,
'x_val': -2.9781042076848507e-09,
'y_val': -3.0370248538247324e-09,
'z_val': 0.693511962890625
}

Also, i noticed that if change sign of w_val, then it almost correct

This new jupyter-notebook https://gist.github.com/KabirovIrik/26f3e09a1087483b01cd2ebd761e44ca

ahmed-elsaharti, 2021-03-17:
Hi @KabirovIrik,
Tbh I didn't exactly understand what you did in the `Include details of what you already did to find answers` section (It does seem to me that you're going down the right path though). 
But in general, (if I remember correctly) to transform points from one coordinate system to another you'll need to create a transformation matrix that expresses the camera frame relative to the world frame:
[image]
This can be created using the drone's(/the camera's) orientation and position returned by `simGetGroundTruthKinematics()`. Since these values are the drone's position and orientation relative to the world coordinate system:
[image]
using something like:
```
local_rotation=client.simGetGroundTruthKinematics().orientation
local_position=client.simGetGroundTruthKinematics().position
RotationMatrix = Rotation.from_quat([local_rotation.x_val,local_rotation.y_val,local_rotation.z_val,local_rotation.w_val])
TranslationMatrix = numpy.array([[local_position.x_val],[local_position.y_val],[local_position.z_val]])
TransformationMatrix = numpy.hstack((RotationMatrix.as_matrix(),TranslationMatrix))
TransformationMatrix = numpy.vstack((TransformationMatrix,numpy.array([0,0,0,1])))
```
once the transformation matrix is formed all you have to do is **dot** multiply (ie: numpy.dot) each point (in the camera coordinate system) with the matrix to transform it into the world coordinate system:
[image]
[image]

(Note that here I'm assuming that the pointcloud is generated in reference to the 'center of the vehicle', the position of-which is returned by `simGetGroundTruthKinematics()`)