microsoft/AirSim #3471: cv capture: RGB Image has no lights

tsaizhenling, 2021-03-16:
## Question
### What's your question?

why does image API capture image without lights? (no shadows are casted)

[image]
[image]

### Include context on what you are trying to achieve

#### Context details

goal: trying to use airsim to capture depth and rgb for deeplearning

OS Windows 10
Unreal engine 4.25.4

set up a custom unreal environment with a scene from marketplace
clicked on play
ran a slightly modified cv_capture.py from examples provided
```
# In settings.json first activate computer vision mode: 
# https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#computer-vision-mode

import setup_path 
import airsim

import pprint
import tempfile
import os
import time

pp = pprint.PrettyPrinter(indent=4)

client = airsim.VehicleClient()

airsim.wait_key('Press any key to get camera parameters')
for camera_id in range(3):
    camera_info = client.simGetCameraInfo(str(camera_id))
    print("CameraInfo %d: %s" % (camera_id, pp.pprint(camera_info)))

airsim.wait_key('Press any key to get images')
tmp_dir = os.path.join("/mnt/c/Users/tsaiz/depthData", "airsim_drone")
print ("Saving images to %s" % tmp_dir)
try:
    for n in range(3):
        os.makedirs(os.path.join(tmp_dir, str(n)))
except OSError:
    if not os.path.isdir(tmp_dir):
        raise

pos = [
        [-2, -1],
        [-2, -2],
        [-2, -3],
        [-2, -4],
        ]
for x in range(len(pos)): # do few times
    #xn = 1 + x*5  # some random number
    client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(pos[x][0], pos[x][1], -1), airsim.to_quaternion(0, 0, 2)), True)
    time.sleep(0.1)

    responses = client.simGetImages([
        airsim.ImageRequest("0", airsim.ImageType.Scene),
        airsim.ImageRequest("1", airsim.ImageType.Scene),
        airsim.ImageRequest("2", airsim.ImageType.Scene)])

    for i, response in enumerate(responses):
        if response.pixels_as_float:
            print("Type %d, size %d, pos %s" % (response.image_type, len(response.image_data_float), pprint.pformat(response.camera_position)))
            airsim.write_pfm(os.path.normpath(os.path.join(tmp_dir, str(x) + "_" + str(i) + '.pfm')), airsim.get_pfm_array(response))
        else:
            print("Type %d, size %d, pos %s" % (response.image_type, len(response.image_data_uint8), pprint.pformat(response.camera_position)))
            airsim.write_file(os.path.normpath(os.path.join(tmp_dir, str(i), str(x) + "_" + str(i) + '.png')), response.image_data_uint8)

    pose = client.simGetVehiclePose()
    pp.pprint(pose)

    time.sleep(3)

# currently reset() doesn't work in CV mode. Below is the workaround
client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(0, 0, 0), airsim.to_quaternion(0, 0, 0)), True)
```

please find unreal project here:
https://drive.google.com/file/d/1Pdoax-zGxH_DZ2DNn8Ps-7u-hS_ZmKMo/view?usp=sharing

### Include details of what you already did to find answers

played with exposure and gamma settings in camera, no improvement

saihv, 2021-03-16:
@tsaizhenling Just FYI, this particular environment you are using uses real time ray tracing, which is an advanced feature that might be causing problems with the airsim cameras. Could you perhaps try with another sample environment from the marketplace to see if the same issue persists?

kestrelm, 2021-03-16:
> @tsaizhenling Just FYI, this particular environment you are using uses real time ray tracing, which is an advanced feature that might be causing problems with the airsim cameras. Could you perhaps try with another sample environment from the marketplace to see if the same issue persists?

Hello,
Thanks for the answer. Is real-time ray-tracing not supported for AirSim cameras? Or is there some other way to get around it?

Thanks

tsaizhenling, 2021-03-16:
tried landscapeMountains and it seems to be okay,

also tried turning off raytracing and now we got the scene and camera capture to match. the quality is a lot lower though :(

[image]

is this a bug?

tsaizhenling, 2021-03-17:
adding
r.RayTracing.SceneCaptures=True to DefaultEngine.ini fixed this issue

closed 2021-03-17 by tsaizhenling