microsoft/AirSim #3479: Does AirSim work with Real-time Ray-tracing? ( Does not seem to work )

kestrelm, 2021-03-17:
## Question
### What's your question?
Hello, we are doing some testing with AirSim using Real-time Ray-tracing. For whatever reason, the captured image from AirSim does not correspond to what we actually see rendered out. Most of the lighting is gone as well as the global illumination.

### Include context on what you are trying to achieve
We are using AirSim for training self-driving cars

#### Context details
We are on UE4.26

### Include details of what you already did to find answers
We tried using GPU Lightmass to bake out the lighting, it is better but the captured results lack shadows compared to what is being rendered.

Thanks

rajat2004, 2021-03-17:
Maybe try https://answers.unrealengine.com/questions/970068/scenecapture2d-has-no-shadows-when-raytrace-enable.html?

kestrelm, 2021-03-17:
How do we use this with AirSim? Any tips will be appreciated. Thanks!

tsaizhenling, 2021-03-17:
adding
`r.RayTracing.SceneCaptures=True` to `DefaultEngine.ini` works!

closed 2021-03-17 by kestrelm