microsoft/AirSim #3481: Can you change MPC vars at runtime with a script?

HoomanRamezani, 2021-03-17:
## Can you change MPC vars at runtime with a script?
Is it possible to change the value of Material Parameter Collection variables during runtime with Python (or any other language)?
I currently have this set up to be a function of keyboard clicks, but I’m hoping to automate things. If it’s not possible to change MPC vars through API calls, is there a recommended way to simulate keyboard clicks in UE4 with a Python script? I’ve found generic ‘simulate keypresses’ packages, but am wondering if there are any specific packages that play nicely with UE4.

 I’m using 
- Linux Ubuntu
- UE4 version 4.24.1
- Python 3.8.5.

rajat2004, 2021-03-18:
Have you added the keyboard bindings in C++ yourself like [this](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L237), or is it like another plugin using Blueprints? In the first case, you could add RPC calls like all the AirSim APIs to call the method, and use them from Python or C++. 
For the latter, options like [pydirectinput](https://github.com/learncodebygaming/pydirectinput) and https://stackoverflow.com/questions/14489013/simulate-python-keypresses-for-controlling-a-game might work, but haven't used them myself.

Sidenote, just to confirm, you're not talking about the distortion params already present in the cameras right? Cause there are APIs for them already

HoomanRamezani, 2021-03-19:
Thanks! Do you have any recommendations for resources on adding your own rpc integration?

rajat2004, 2021-03-20:
For examples on how to add a new API, you can see [this commit](https://github.com/Microsoft/AirSim/commit/f0e83c29e7685e1021185e3c95bfdaffb6cb85dc), and various other PRs like #2834 or https://github.com/microsoft/AirSim/pull/2867. Best way to get some help if stuck somewhere might be to push to a branch and post here or maybe open a draft PR

jonyMarino, 2021-03-23:
Hi @HoomanRamezani! and welcome! Out of curiosity: Why are you interested in changing MPC params during simulation?