microsoft/AirSim #3482: Weather and time parameters not working appearing when in Computer Vision mode

HoomanRamezani, 2021-03-17:
## Question
### Weather and time parameters not working appearing when in Computer Vision mode
I am trying to change weather and time parameters when recording in computer vision mode. In simulation mode the code works as expected, changing weather and time as specified. When using computer vision the ‘fog’ and ‘dust’ modes work, but the ‘rain’ and ‘snow’ weather conditions don’t appear, and time changes are not made. I’m wondering if this has something to do with particle simulation.

This code I’m using is:
```
self.client.simSetWeatherParameter(airsim.WeatherParameter.Dust, magnitude)
self.client.simSetTimeOfDay(is_enabled=is_enabled, start_datetime=time_to_set)
```
Does anyone know what’s going on?

#### Context details
I’m using
- Linux Ubuntu
- UE4 version 4.24.1
- Python 3.8.5.

zimmy87, 2021-04-08:
I am not able to repro this with Ubuntu and UE 4.25 as of commit [6d20663](https://github.com/microsoft/AirSim/commit/6d20663e0818448613be0332009b02bd6341589e). Please try to repro this with the latest sync of master and then with UE 4.25. If this issue still repros at that point, please let us know.

HoomanRamezani, 2021-04-12:
I have further investigated the issue, there appears to be a 'particle box' created around the actor in UE4 that displays rain/snow particles in a small box a few meters around the actor. This box does not seem to move with the computer vision camera recording so once it's moved away from the actor there are no rain/snow particle effects. I am looking to change this in order to ensure that the rain/snow particles appear everywhere in the sim so they appear in the recordings.