microsoft/AirSim #3484: Can't build Blocks project for Unreal on Linux

petergerten, 2021-03-18:
Engine: UE4.25.4-release
OS: Linux (Ubuntu 20.04)
AirSim: master

When trying to build the Blocks project on Linux I get the following Errors:
---------------
```
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/AirsimLevelStreaming.cpp:1:9: error: #pragma once in main file [-Werror,-Wpragma-once-outside-header]
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/AirsimLevelStreaming.cpp:18:17: error: use of undeclared identifier 'FPackageName'
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/AirsimLevelStreaming.cpp:26:35: error: use of undeclared identifier 'FPackageName'
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/AirsimLevelStreaming.cpp:27:30: error: use of undeclared identifier 'FPackageName'
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/TextureShuffleActor.cpp:13:66: error: member access into incomplete type 'UStaticMeshComponent'
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/TextureShuffleActor.cpp:26:43: error: incomplete type 'UMaterialInstanceDynamic' named in nested name specifier
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/TextureShuffleActor.cpp:29:27: error: member access into incomplete type 'UStaticMeshComponent'
   ActionThread.ActionDebugOutput: /home/unreal/Blocks/Plugins/AirSim/Source/TextureShuffleActor.cpp:32:39: error: member access into incomplete type 'UMaterialInstanceDynamic'
   ActionThread.ActionDebugOutput: /UnrealEngine/Engine/Source/Runtime/Engine/Classes/GameFramework/Actor.h:3010:32: error: incomplete type 'UStaticMeshComponent' named in nested name specifier
   ActionThread.ActionDebugOutput: /UnrealEngine/Engine/Source/Runtime/Engine/Classes/GameFramework/Actor.h:3010:33: error: incomplete definition of type 'UStaticMeshComponent'

```
---------------
To fix this I have to do the following:
- remove pragma dependency
- add   #include "Misc/PackageName.h"
- add   #include "Components/StaticMeshComponent.h"
- add   #include "Materials/MaterialInstanceDynamic.h"

Am I doing something wrong here?

On a side note: any reason why there is no integration test via github action for the Unreal builds? I only see the Unity builds there.

rajat2004, 2021-03-18:
That's pretty strange, I haven't had these errors myself, using Ubuntu 18.04 & UE 4.25.3. The `pragma` error definitely should've occurred at least

As to the CI, Azure Pipelines does the UE4 build on Ubuntu and Windows. The GitHub Actions Unity build only builds the wrapper, and doesn't actually have Unity with it for a full compilation

petergerten, 2021-03-27:
I am not sure how to approach this - I would be happy to share a minimal example but even that is quite big and takes time considering the I am building UE4.
I could reproduce this consistently with Ubuntu 20.04 and UE4.25.4-release.

As written above, adding those dependencies fixes the build  - do you have any hints why they should **not** be required?

rajat2004, 2021-03-27:
I myself feel that these dependencies should be required, the errors should have occurred. Checked the 4.25.4 release notes but nothing seemed to pop out. Anyways, there shouldn't be any problem in adding these files

sethmnielsen, 2021-04-05:
I had the same problems today - on Arch Linux, UE 4.25.4. @petergerten's fixes worked for me.

However, I didn't have any errors when building the Blocks project for the editor; the errors occurred while trying to **package** Blocks (i.e. for use as a stand-alone binary). @rajat2004 maybe you'd get the errors if you tried to package the project as well?

petergerten, 2021-04-27:
@sethmnielsen indeed, I get the error during packaging.

zimmy87, 2022-01-14:
Hi @petergerten, are you still experiencing this error?