microsoft/AirSim #3487: MotionBlurAmount not changing the amount of blur in images

ptueller, 2021-03-19:
## Question
### What's your question?
What is the MotionBlurAmount parameter specifically blurring?

### Include context on what you are trying to achieve
I am trying to simulate changing shutter speed on the downward facing camera on the multirotor platform. Ideally I would like to see blur increase as I move the multirotor faster. However, no matter what I set the parameter 'MotionBlurAmount' as, I observe no change in the images that I gather.

#### Context details
Windows 10, AirSim 1.4.0, Python 3.7.7, Unreal 4.24.3
Here is my settings.json file:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.1,
    "Cameras": [
        { "CameraName": "3", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "Width": 1280,
        "Height": 720,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 1.0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }
}

### Include details of what you already did to find answers
I've tried to adjust blur settings directly within Unreal as well as within AirSim, but nothing is changing. This link says that AirSim uses Render Targets, which do not support motion blur: https://midair.ulg.ac.be/faq.html. If that is the case, what is the MotionBlurAmount parameter for? Does it only render blur for objects within the environment that are in motion, and does not render blur based on the motion of the camera?

jfswitzer, 2021-04-01:
Any updates on this?

ztysdu, 2022-10-20:
Hello, have you solved this problem? I am also confused of how to set motion blur.

ptueller, 2022-10-20:
Hello, unfortunately I can't remember exactly what I did to achieve it, though I did get it to somehow work. Sorry I can't be more helpful, I've moved on to another project and group and the code I used has been archived. I do remember though that it did not end up working how I was expecting it to work, in that the motion blur was not caused by the motion of the vehicle, and it was instead what seemed to be a post-processing on the image. I was trying to replicate shutter speed and the simulator isn't so detailed that it replicates that.

Leon-xc, 2022-12-05:
@ztysdu  Have you solved this problem ? I encounter the same problem.