microsoft/AirSim #3493: WSL2 AirSim Connection error

Rohan138, 2021-03-20:
## client.confirmConnection() error on WSL2 but not on Windows host
I'm trying to connect to AirSim and Unreal running on my Windows host through a python script in WSL2. I can connect successfully and run the script from my Windows host, but the MultirotorClient() fails to connect on WSL2.

### Context:

OS: Windows Build 21332; WSL2: Ubuntu 18.04 LTS
Script: [multi_agent_drone.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/multi_agent_drone.py)
AirSim v1.4.0; UE4.26
Python 3.8.5 on both Windows and WSL2

### Error:
python multi_agent_drone.py
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
Traceback (most recent call last):
  File "multi_agent_drone.py", line 33, in <module>
    client.confirmConnection()
      File "/home/rohan/miniconda3/lib/python3.8/site-packages/airsim/client.py", line 139, in confirmConnection
    if self.ping():
      File "/home/rohan/miniconda3/lib/python3.8/site-packages/airsim/client.py", line 44, in ping
    return self.client.call("ping")
      File "/home/rohan/miniconda3/lib/python3.8/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
      File "/home/rohan/miniconda3/lib/python3.8/site-packages/msgpackrpc/future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit

### Fixes tried:
Disabled Windows Defender Firewall
Passed in ip addresses '0.0.0.0' and '127.0.0.1' and port 41451 explicitly to MultirotorClient()
Copied settings.json to /home/username/Documents/AirSim on WSL2

Any ideas on what's going wrong here?

closed 2021-03-20 by Rohan138

abdo-ameen, 2021-06-30:
can you please explain what did you do accurately?

Rohan138, 2021-06-30:
In bash (.bashrc to make permanent):

export WSL_HOST_IP=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}')

In your script:

HOST = '127.0.0.1'  # Standard loopback interface address (localhost)
from platform import uname
if 'linux' in uname().system.lower() and 'microsoft' in uname().release.lower(): # In WSL2
    if 'WSL_HOST_IP' in os.environ:
        HOST = os.environ['WSL_HOST_IP']
        print("Using WSL2 Host IP address: ", HOST)
client = AirSim.multiiRotorClient(ip=HOST)

abdo-ameen, 2021-06-30:
That works 👍 
The key is to give the `WSL_HOST_IP` as an argument to the client
`client = airsim.MultirotorClient(ip=HOST)`
As a precaution I added the `WSL_HOST_IP` to my `setting.json`
`"LocalHostIp": "172.23.32.1",`
Thank you so much for the help 👌

Rohan138, 2021-06-30:
You're welcome, but the WSL_HOST_IP changes every time you restart your computer.

abdo-ameen, 2021-06-30:
For sure, and your code does that automatically in the script, making it more convenient. Thank again.