microsoft/AirSim #3498: angular random walk noise calculation in imu addnoise

juwangvsu, 2021-03-21:
## Question
### What's your question?

In ImuSimple.hpp, the imu random walk noise is calculated by:
 real_T gyro_sigma_arw = params_.gyro.arw / sqrt_dt;
is this right? should not it be arw multiple with the sqrt_dt?

### Include context on what you are trying to achieve

I am trying to use the IMU data to estimate odometry, but the noise level seems way to high.

#### Context details

airsim 1.3.0 on windows 10. Unreal 4.25

setting on IMU:
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
### Include details of what you already did to find answers

rajat2004, 2021-03-23:
I had a look at the reference, and seems that the code is wrong as you mentioned

Page 11, https://www.cl.cam.ac.uk/techreports/UCAM-CL-TR-696.pdf
[image]

Maybe open a PR and the maintainers can decide

jwvsu, 2021-03-26:
also there are some related issues with frame_id in ros published topic. airsim_ros_wrapper node published imu topic with the frame id set to vehicle name, which is fixed to world frame. I think the frame_id should be the body link, such as SimpleFlight/odom_local_ned.
However the imu message contain orientation , linear_acc, and gyro_velocity. They seems assume different frame_id. the orientation in rightfullly under vehichle name frame, but the other two should be in body frame.