microsoft/AirSim #3515: How can we read in the input data of Remote Controller when the drone is in the API mode?

AlexanderZ3, 2021-03-23:
## Question
### How can we read in the input of Remote Controller when the drone is in the API mode?  
In other words, Can we activate the RC mode and API mode simultaneously?

I need to simulate the situation that, the drone is flying by following the program, but human operators can intervene and take the control authority at any time. Can this function be achieved in Airsim? 

Since I found that when I activate the API mode, it seems impossible to read the data of the Remote controller. Maybe someone has some clues, thanks!

ahmed-elsaharti, 2021-03-23:
The solution I'm about to describe is probably the least elegant one, but the following is what I did when I tried to simulate the exact same scenario a couple of years ago:
Assuming you're utilizing the Python API calls, you could use a library to constantly monitor the controller (such as [Xinput](https://pypi.org/project/XInput-Python/) but i remember having to make some changes to it, idr why) and call `enableApiControl(False)` when the sticks are moved. 
The main issue with this was that I could not use the default .join() blocking call (in whatever API control call I'm making such as moveto... etc) since it wouldn't my code run the 'controller being used' check during the movement. What I did was I created my own internal function that loops internally till the movement command is done (using delays and whatnot...ie: not a clean way at all. If you're doing a movetoposition command I recommend just checking the status of the task by polling the position of the drone within the loop to know if it reached) while checking the state of the controller.

This should technically work...I think....

AlexanderZ3, 2021-03-24:
@ahmed-elsaharti Thanks! l will take a try!

closed 2021-03-24 by jonyMarino