microsoft/AirSim #3530: AirSim ROS wrapper uses settings.json at default path, regardless of settings given to AirSim

ercanburak, 2021-03-26:
## Bug report 

- AirSim Version/#commit: v1.4.0 - Linux / 6d20663e0818448613be0332009b02bd6341589e
- UE/Unity version: UE4 version 4.25.1
- autopilot version: N/A
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

AirSim has the ability to load settings using 4 different ways, as described in [docs](https://github.com/microsoft/AirSim/blob/master/docs/settings.md). For example, one can give the path to a specific settings file with the `--settings` command line argument, instead of loading the settings from `~/Documents/AirSim/settings.json`. However, while AirSim can load the correct settings as described in the docs, the AirSim ROS wrapper always uses settings from the file at default path, i.e. `~/Documents/AirSim/settings.json`. 

Therefore, if one starts an Unreal Engine environment binary with different settings (other than the one at the default path), this gives result to an inconsistency between the real settings used by AirSim, and the settings AirSim ROS wrapper thinks that are used. This inconsistency can cause many more issues on the ROS side. As an example, please read the settings and reproducing steps below.

### Settings

I use two different setting files to demonstrate the issue, which I share [in this gist](https://gist.github.com/ercanburak/b7f0359f490aff28ffdc715efe422509). First, we have a `my_settings.json` file that we give to AirSim, with a command like `/home/bercan/Downloads/LandscapeMountains/LinuxNoEditor/LandscapeMountains.sh --settings '/home/bercan/Documents/AirSim/my_settings.json'`. Second, we have a `settings.json` file with the default name, which we put at the default path (under `~/Documents/AirSim` on Linux systems). Both of these setting files include a camera, and the only difference they have is the camera name, with `my_settings.json` having `MyCamera1` as the name, and `settings.json` having `Camera1`.

### How can the issue be reproduced?

1. Have two different setting files ready, with inconsistent settings, with first having the default name `settings.json` and second having a different name, such as the ones that I share [in this gist](https://gist.github.com/ercanburak/b7f0359f490aff28ffdc715efe422509), and put them on default folder (under `~/Documents/AirSim` on Linux systems).
2. Run AirSim by giving the path to the second settings file (by using the first option in the [doc](https://github.com/microsoft/AirSim/blob/master/docs/settings.md)), with a command like `/home/bercan/Downloads/LandscapeMountains/LinuxNoEditor/LandscapeMountains.sh --settings '/home/bercan/Documents/AirSim/my_settings.json'`, and observe that the correct settings at 'my_settings.json' are loaded, as in the screenshot below.
[image]
3. Have your AirSim ROS wrapper built and sourced as described in [docs](https://microsoft.github.io/AirSim/airsim_ros_pkgs/), and launch the AirSim ROS wrapper node with the command `roslaunch airsim_ros_pkgs airsim_node.launch`
4. By using a command like `rostopic list`, observe that the AirSim ROS wrapper tries to publish camera images from a camera named `Camera1` as in `settings.json`, instead of `MyCamera1` as in `my_settings.json`.
5. Also, by using a command like `rostopic echo /airsim_node/MyCar/Camera1/Scene/compressed`, observe that the camera images are not actually published.

rajat2004, 2021-03-28:
Hey, could you test my branch at https://github.com/rajat2004/AirSim/tree/ros-settings, I added an API to get the settings text which Airsim uses, and then the ROS wrapper calls the API to get the same text. This should remove the problem of different settings files, since only the Airsim binary or running in Editor will read the file. Needs some more cleanup, will probably open a PR after that

ercanburak, 2021-03-31:
Hey @rajat2004, thanks for adding the API, I was also thinking that this approach should be the one to solve this issue. I have fetched and built your branch. However, I'm not sure how to test it since I was using the Landscape Mountains environment that I have downloaded in binary format, under Linux. I normally use the command `/home/bercan/Downloads/LandscapeMountains/LinuxNoEditor/LandscapeMountains.sh --settings '/home/bercan/Documents/AirSim/my_settings.json'` to run the simulation. How should I proceed, do you have any suggestions?

rajat2004, 2021-03-31:
You can try building Airsim from source and using the Blocks environment, but if you haven't done that before then it will require downloading and building UE4, which takes time and space. I could have built a binary for Linux, however I'm travelling currently and don't have my laptop with me. This is the reason why I wanted to have the ability to download binaries from the Azure CI, it already builds and zips it up as well, I think only thing left is to upload it as an artifact. I can't myself test and add this feature since that requires access to the Azure DevOps stuff, but the maintainers should be able to.

ercanburak, 2021-04-01:
Actually I have built UE4 and Airsim from source (including your changes). But I am not sure how to specify a custom settings file for Blocks environment, since I was running that environment under Unreal Editor with a command like `/home/bercan/Unreal4.24/Engine/Binaries/Linux/UE4Editor /home/bercan/AirSim/Unreal/Environments/Blocks/Blocks.uproject`. I do not know where to find (or how to generate) the `./Blocks.sh` script mentioned in the [docs](https://github.com/microsoft/AirSim/blob/master/docs/settings.md), to give a settings file to.

rajat2004, 2021-04-01:
The arguments can be passed in the same manner, like `/home/bercan/Unreal4.24/Engine/Binaries/Linux/UE4Editor /home/bercan/AirSim/Unreal/Environments/Blocks/Blocks.uproject --settings ...`

PR #3536 (2021-04-13): ROS: Use the same settings as AirSim 

Fixes: #3530    
Fixes: #3464
Fixes: #2415
Others: #2365, #2009

## About

This adds an API to get the settings text which is being used by AirSim. The API is then used by the ROS wrapper to fetch the settings text and uses it to instantiate its own settings.
This felt like the cleaner way to fix the problem at its root, it'll avoid requiring the same logic to search for the settings.json file, and now doesn't matter if Airsim and the ROS wrapper are running in different machines. 

Any suggestions on a better way to implement this would be great!

## How [... cut]
closed 2021-04-13 by jonyMarino