microsoft/AirSim #3533: Python simGetImages: Attribute 'to_msgpack' missing

tkazik, 2021-03-26:
## Bug report 

- AirSim Version/#commit: latest / 6d20663e0818448613be0332009b02bd6341589e
- UE/Unity version:  4.26.1
- OS Version: Ubuntu 18.04 **and** 20.04

### What's the issue you encountered?

Running the superbasic tutorial `python3 hello_drone.py`: The script crashes while getting the images.

[image]

### Settings

=> default (installed `pip install msgpack-rpc-python`)

### How can the issue be reproduced?

every time on Ubuntu 18.04 and 20.04

### Include full error message in text form

```shell
    sock.send_message(message, callback)
  File "/home/xxx/anaconda3/envs/myCVLpy3/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "msgpack/_packer.pyx", line 284, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 290, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 287, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 234, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 277, in msgpack._packer.Packer._pack
  File "/home/xxx/anaconda3/envs/myCVLpy3/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 12, in <lambda>
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
AttributeError: 'function' object has no attribute 'to_msgpack'
```
Thx!

hpsauce82, 2021-03-27:
I get the same error on Windows 10 running hello_car.py

tkazik, 2021-03-30:
This is likely related to an issue with [msgpack-rpc-python](https://github.com/msgpack-rpc/msgpack-rpc-python), which is kind of old and does not work with tornado >= v5 (the current tornado version is 6.1). Probably because the msgpack-rpc-python package is not really maintained anymore, see https://github.com/msgpack-rpc/msgpack-rpc-python/issues/16. Anyone else has another take?

mmorphew, 2021-04-01:
The solution suggested in #3553 fixed this error for me. To reiterate that fix, navigate to `AirSim/PythonClient/airsim/types.py`, and within that file, comment out both the entire `_ImageType` class, as well as the redefinition of the `ImageType` class that immediately comes after it. This reverts `ImageType` to use its first definition, in which each image type maps to an integer. After this step is completed, replace the instances of `ImageType` referenced in `hello_drone.py` with their corresponding integers for each `ImageRequest`. (For example, `airsim.ImageRequest("0", 0)`. The script should run successfully and output the images to the specified directory in the script. I have tested this with all image types on Ubuntu 20.04 and have been able to get successful outputs for all of them.

Roopesh-Bharatwaj-K-R, 2021-04-01:
Hi guys , how you all managed to connect with Python Environment , i tried a lot but i could not able to connect it

ahmed-elsaharti, 2021-04-01:
@Roopesh-Bharatwaj-K-R [This guide](https://microsoft.github.io/AirSim/apis/) could be a good place to start

ahmed-elsaharti, 2021-04-04:
Also, can you guys try the solution suggested in [this comment](https://github.com/microsoft/AirSim/issues/3553#issuecomment-812933387) to see if its an issue with the python modules installed

PR #3594 (2021-04-15): remove duplicate definition of ImageType class

Fixes: #3533
Fixes: #3553
Related to: #3550 

## About
This PR addresses an error where accessing fields in the ImageType class would generate an error: "AttributeError: 'function' object has no attribute 'to_msgpack'". The fix is to remove a duplicate definition of the ImageType class in types.py

## How Has This Been Tested?
hello_drone.py script was run and confirmed to complete execution without error

## Screenshots (if appropriate):
closed 2021-04-15 by jonyMarino