microsoft/AirSim #3537: Aquire Lidar intensity

dongfeng12, 2021-03-29:
As is mentioned in issue #2290, we only have coordinate info for Lidar data. 

In [Carla](https://carla.readthedocs.io/en/0.9.11/ref_sensors/#lidar-sensor)'s implementation, intensity is calculated by distance and atteunation rate, which is not consistent with the reality.  Is it realizable to get more realistic intensity data, e.g. from texture of UE 
object or the normal vector of the collision surface.

wouter-heerwegh, 2021-11-15:
Hi @dongfeng12,

Have you managed to do this? I'm looking for the same thing. I plan on havimg a look at this and starting a draft pull request to implement something.

I do not have a lot of experience with Unreal engine, so all the help is appreciated.

wouter-heerwegh, 2021-12-15:
I've been looking at this for a while, and there does not seem to be a very efficient way of doing this in unreal engine. I'm able to get texture info at the specific point I hit, but no color data.

If anyone else finds a nice way to implement this, feel free to add extra information.

andrewjong, 2024-01-27:
It seems this project has extended AirSim's LiDAR to include intensity
https://github.com/Cosys-Lab/Cosys-AirSim