microsoft/AirSim #3540: Compiling C++ Code on Linux

mmuetert, 2021-03-29:
## Question
### What's your question?

I am trying to compile custom C++ code to use with AirSim. As a test I copied the source code and CMakeLists.txt files to new directories called HelloDroneTest. I updated the project name and the directory in the CMakeLists.txt file as appropriate. I ran cmake and it seemed to finish without a problem. I then ran make and recieved the following error:

 make
[ 50%] Linking CXX executable output/bin/HelloDroneTest
/usr/bin/ld: cannot find -lAirLib
/usr/bin/ld: cannot find -lrpc
collect2: error: ld returned 1 exit status
CMakeFiles/HelloDroneTest.dir/build.make:94: recipe for target 'output/bin/HelloDroneTest' failed
make[2]: *** [output/bin/HelloDroneTest] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/HelloDroneTest.dir/all' failed
make[1]: *** [CMakeFiles/HelloDroneTest.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

It seems it can't find the library, but this file should be the one used to build HelloDrone in the first place and I didn't change that part of the file. Does anyone know how to fix this?

### Include context on what you are trying to achieve

#### Context details

[CMakeLists.txt](https://github.com/microsoft/AirSim/files/6224038/CMakeLists.txt)
OS: Ububtu 18.4
Airsim Version: Newest
Python: 3.6
default json

### Include details of what you already did to find answers

jonyMarino, 2021-04-15:
Hi @mmuetert! Try following the tutorials in this issue: #2991.
But that error appears when you are not pointing to the correct build folder.
Let me know if that help

zimmy87, 2022-01-14:
Hi @mmuetert, are you still experiencing this issue?