microsoft/AirSim #3550: Build Error in Hello Drone

Roopesh-Bharatwaj-K-R, 2021-04-01:
Build Error

we are trying to run the hello drone but still problem persist after installing all packages. 

 trying to run the hello drone in landscape mountains, pops the error this was one scenario. 
next is  when is insert the joystick control it runs and it pops error showing do you want to proceed with the previous successful build , if yes ...joystick control is enabled with drone .
is this hello drone program automatically allows to fly or this only allows to use aid with joystick control ?

 Context details
windows-10, Python 3.7, Unreal- 4.26 
[image]

richtong, 2021-04-01:
We are trying to run your exact scenario. I feel super dense. The READMEs are scattered a bit so hard to know where to start. But at https://microsoft.github.io/AirSim/use_precompiled/

We have windows 10. We downloaded the binaries from the release page and unzipped it. The README says look for https://github.com/Microsoft/AirSim/releases

we unzipped the files. And it looks like an image of the GitHub repo and then it says to “ run the run.bat” I don’t see any file like this in that in the .zip file. So I think we are looking in the wrong place for installation. What’s the correct instruction?

Roopesh-Bharatwaj-K-R, 2021-04-01:
Hi RichTong, To be Honest, MS-Airsim,  instructions are not that much clear and it vague and the Video explanation available on the you tube are pretty old and faced same problem with installation . Some how managed to build and run and now completedly freezed dono where to proceed - our integration from visual studio code to python -Anaconda environment is very challenging one , we followed the instruction but still we could not ....

Tried Implementing with AZURE : 
its again staring from scratch creation of new VM, then - connecting with ssh to visual studio code,, downloading all packaged in VM ... finally it was running on different environment but here also we want to implement the RL algorithms and we struck where to start over and code in which part....

ahmed-elsaharti, 2021-04-01:
@richtong double clicking the executable should launch the binary:
[image]
the `run.bat` suggestion is more of a 'create a run.bat file with `start CityEnviron -ResX=640 -ResY=480 -windowed` in it if you want to run in windowed mode directly' thing

@Roopesh-Bharatwaj-K-R you can just install the `msgpack-rpc-python` and `airsim` libraries by running:
`pip install msgpack-rpc-python`
and
`pip install airsim`
in your terminal (assuming you've got python and pip installed)
then you can go (cd) to the directory of the script you want to run (for example, for hello_drone we'd have to start a terminal at the Airsim/Pythonclient/multirotor directory) and run the script using 
`python scriptname.py` (eg: `python hello_drone.py`)

Roopesh-Bharatwaj-K-R, 2021-04-02:
@ahmed-elsaharti  Thank You very Much , 

Actually, we tried running this and but still face error,  i'm attaching the image of that , please look over and give us some idea and suggestions. Thanks Again.
[image]

ahmed-elsaharti, 2021-04-02:
@Roopesh-Bharatwaj-K-R This error suggests that script cant reach an airsim instance. Are you sure you are running an airsim simulation/binary while testing the script?

Roopesh-Bharatwaj-K-R, 2021-04-02:
@ahmed-elsaharti , Thanks...It works well now,  but the drone moves upwards and its bit erratic ,  Actually where do we need to input the required code for object detection and autonomous driving(pathing). Is it going to be in the computer vision folder or we can code it directly into hello drone.py?.

Roopesh-Bharatwaj-K-R, 2021-04-02:
@ahmed-elsaharti, I've also been having problem with running the code. it Always doesn't build the current instance because there are errors in myclass.h, I haven't even tempered with the file. Attached is the error, can you please help me out. Thanks in Advance.
[image]

ahmed-elsaharti, 2021-04-02:
@Roopesh-Bharatwaj-K-R You could create a python script using your IDE of choice and run it either using the IDE or through terminal as long as your script imports airsim just like the demo scripts (hello_drone, etc) do. I generally recommend doing that rather than trying to run python through visual studio 2019

Roopesh-Bharatwaj-K-R, 2021-04-03:
@ahmed-elsaharti , At first this was working well but then i tried to clone some files from github(not really sure the cause). Had to reinstall python and trying to run hello_drone.py again didnt work
[image]

Roopesh-Bharatwaj-K-R, 2021-04-03:
@ahmed-elsaharti  Thank You, Uninstalling Python environment from scratch and  reinstalling.

Roopesh-Bharatwaj-K-R, 2021-04-03:
@ahmed-elsaharti  its working now !

Roopesh-Bharatwaj-K-R, 2021-04-03:
@ahmed-elsaharti  Hi, we tried Running Hello drone, every thing was fine and in mid way we reached with an unexpected Attribute Error, Thanks.
[image]

ahmed-elsaharti, 2021-04-03:
Hi @Roopesh-Bharatwaj-K-R , can you try [this solution](https://github.com/microsoft/AirSim/issues/3553#issuecomment-812933387) and let us know if it works?

Roopesh-Bharatwaj-K-R, 2021-04-04:
Hi, @ahmed-elsaharti  We followed the Solution provided from the link , of uninstalling of airsim/ msg-rpc python package , we did it but still there error seems to be persist , 
we even upgraded msg 
[image]
pack to latest and tornado as well.(It seems different packages require different versions of tornado to work. msgpack-rpc-python for example requires lower than version 5 of tornado, while notebook 6.1.4 and distributed 2.30.1 require version 6 and above.) I ignored the others and just downloaded version 4.5.3 since we weren't really using those modules for the hello_drone. It started working but didn't take pictures at the end. And so comes the error. again, Please Give some suggestions to proceed further, Thanks.

ahmed-elsaharti, 2021-04-04:
@Roopesh-Bharatwaj-K-R try using `pip list` as suggested in the comment then uninstall `msgpack` and `msgpack-python`.

ahmed-elsaharti, 2021-04-04:
After that reinstall the two needed packages again.
Based on the error message it seems like you still have msgpack installed

ahmed-elsaharti, 2021-04-04:
1. The reproducing the error:
[image]

2. Uninstalling `msgpack` and `msgpack-python`
[image]
[image]

3. Reinstalling airsim and deps:
[image]

4. Error message gone:
[image]
[image]

Roopesh-Bharatwaj-K-R, 2021-04-04:
Hi, @ahmed-elsaharti , Thanks for the clear image explanation but  to be Honest we  uninstalled the Both Msgpack, msgpack-python and looked for the Pip list, and we reinstalled  them both, and tried couple of times but still the problem does not move out. Thanks Again.

Attaching the image :  For Uninstalling and Installing .....
[image]
Image : Error at the press a key to take  Image 
[image]

ahmed-elsaharti, 2021-04-04:
@Roopesh-Bharatwaj-K-R Hmmm...thats odd
can you list down all the packages shown by `pip list`?

ahmed-elsaharti, 2021-04-04:
Also, maybe try creating a fresh new conda environment and install ONLY the airsim package? (don't upgrade anything since upgrading msgpack is what causes one of the issues)

Roopesh-Bharatwaj-K-R, 2021-04-04:
> @Roopesh-Bharatwaj-K-R Hmmm...thats odd
> can you list down all the packages and versions shown by `pip list`?

[image]
[image]
[image]
[image]

ahmed-elsaharti, 2021-04-04:
Try a new conda environment and see if that works. That could help us understand if it's a package that's interfering.

Roopesh-Bharatwaj-K-R, 2021-04-04:
> Try a new conda environment and see if that works. That could help us understand if it's a package that's interfering.

Created New Conda Env and installed Airsim, But still same error 
[image]
[image]

ahmed-elsaharti, 2021-04-04:
@Roopesh-Bharatwaj-K-R did you change hello_drone.py?

Roopesh-Bharatwaj-K-R, 2021-04-04:
> @Roopesh-Bharatwaj-K-R did you change hello_drone.py?

yes, i have modified ..

ahmed-elsaharti, 2021-04-04:
> > @Roopesh-Bharatwaj-K-R did you change hello_drone.py?
> 
> yes, i have modified ..

try changing it back.

Roopesh-Bharatwaj-K-R, 2021-04-04:
> > > @Roopesh-Bharatwaj-K-R did you change hello_drone.py?
> > 
> > 
> > yes, i have modified ..
> 
> try changing it back.

Now re modifying to original drone code --- Previous error Occurs...
[image]

Roopesh-Bharatwaj-K-R, 2021-04-04:
@ahmed-elsaharti , What could be the possible solution to resolve this and have tried entire suggested approaches, but Error Persist.
Please Help , Thank You.

Roopesh-Bharatwaj-K-R, 2021-04-07:
@ahmed-elsaharti  Hi, awaiting your kind response ... to tackle this error issue,  Thanks !

ahmed-elsaharti, 2021-04-08:
Hi @Roopesh-Bharatwaj-K-R, I'm unable to reproduce the error which is why I'm not able to debug it.
In my testing the hello_drone script has always run just fine. However, I havent tested with the latest commit. 

What environment are you testing this on? Is it the blocks environment from the latest Airsim commit? If thats the case, try testing with the precompiled blocks binary in the [releases section](https://github.com/microsoft/AirSim/releases). 

If the issue still occurs with the precompiled binary then I'm almost certain that this is happening due to the way you've got python set up on your machine or because of the packages you have installed.

If the script works fine using the binary then this could be an issue caused by a recent commit. If thats the case do let me know and we could try to track down the commit causing this.

Roopesh-Bharatwaj-K-R, 2021-04-08:
Hi @ahmed-elsaharti, we changed the airsim entire to old airsim which we had in another computer ..and new one was cloned in github very recently...i think the new version caused us this error----(we tried both City Landscape and Landscape Mountains, both had the errors on newly cloned airsim even after installing and uninstalling packages). Thank You very much for your support.

jonyMarino, 2021-04-15:
Hi!, this has become a long thread. Is there anything open after PR #3594 from David?

ilopezfr, 2021-05-29:
Has there been any progress in this bug? I'm experiencing the same error and tried all the proposed workarounds in this thread w/o success. Thanks!

jonyMarino, 2021-06-01:
Hi @ilopezfr! Which commit are you using?

closed 2021-07-23 by jonyMarino