microsoft/AirSim #3551: AirSim used in ue4.26 causes water position offset

yizhendongfeng, 2021-04-01:
## Bug report 
- AirSim Version/#commit:V1.4
- UE/Unity version: 4.26
- autopilot version: PX4 fmu-v2 HITL
- OS Version: Win10 64bit

### What's the issue you encountered?
I put some water actor from water plugin to the landscape with GameMode Override set to none, look like this.
[image]

But when i changed the GameMode Override to AirSimGameMode and pressed the play button, water actor got some offset. 
[image]

It seems that water mesh moves to player start position when pressing the play button. Then I set player start location to (0,0,0), water mesh is in the origin place without offset. 
What may cause this problem and how to solve it? Many thanks. 

### Settings
I'm using default HITL settings with px4 vehicle.

### How can the issue be reproduced?
1. create a landscape, add some water actor and a player start actor, then change GameMode Override to AirSimGameMode.
2. Press the play button, you should see this problem.

jonyMarino, 2021-04-16:
Hi @yizhendongfeng, and welcome! Not sure, but at initialization, AirSim changes The world origin to the player start position. But that is the intended behavior. Maybe there is a bug in the water plugin. You can do as a workaround to add the player start location as an offset to the water.

jonyMarino, 2021-07-23:
Also you can set the vehicle at the world's origin and then do a setVehiclePosition() at start

jonyMarino, 2021-07-23:
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

HSC-1, 2022-05-11:
Hello, this is a closed issue, but it seems to be a similar issue, can you please reopen it?
The offset of the buoyancy simulation seems to happen regardless of the level currently loaded.

Other game modes work normally, but the airsim game mode doesn't work well. The same problem occurs even if the world origin is the player start.

**Other game modes**
[image]

**Airsim game mode**

level position
[image]
buoyancy position 
[image]

Used Asset 
[Waterline PRO](https://www.unrealengine.com/marketplace/ko/product/waterline)

@jonyMarino

jonyMarino, 2022-05-11:
@HSC-1, can you try commenting [this line](https://github.com/ahmed-elsaharti/AirSim/blob/17d2b5d1067f6923519d7e70018fc039ec8d8130/Unreal/Plugins/AirSim/Source/SimMode/SimModeBase.cpp#L111), and following the steps to build AirSim from source (build, upgrade_from_git etc.)?

HSC-1, 2022-05-12:
Thanks, it works fine. 
#4517 
But I have a problem similar to the issue here. 

The buoyancy seems to be applied
https://user-images.githubusercontent.com/76798501/167986364-6e56675a-3d76-4c94-be57-2e640bf63fcc.mp4

But the bubble effect is not applied.

**Other game modes**

https://user-images.githubusercontent.com/76798501/167987376-05459419-120c-4cc9-99b8-7c75cf827bfd.mp4

**Airsim mode**

https://user-images.githubusercontent.com/76798501/167987421-2875cff5-f322-4193-b94f-11531a213627.mp4