microsoft/AirSim #3553: Running hello_car.py gives AttributeError: 'function' object has no attribute 'to_msgpack'

zkytony, 2021-04-01:
## Bug report 

- AirSim Version/#commit:  6d20663e08184 
- UE/Unity version:  4.25
- autopilot version:
- OS Version:  Ubuntu 18.04

### What's the issue you encountered?

I just got started and installed AirSim and the Blocks environment seems to work fine.
When I tried running the `hello_car.py` example, however, I get an AttributeError. The full output is below. Here is the screenshot of the simulator running with the error message
[image]

I suspect it's because in the following piece of code in `hello_car.py`:
```python
    responses = client.simGetImages([
        airsim.ImageRequest("0", airsim.ImageType.DepthVis),  #depth visualization image
        airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True), #depth in perspective projection
        airsim.ImageRequest("1", airsim.ImageType.Scene), #scene vision image in png format
        airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGB array
```
`airsim.ImageType.DepthVis` is a function, and thus does not have `to_msgpack` attribute.
What should it be instead? 

One thing I noticed is in `types.py`, there are two `ImageType` classes.  The second one gets used and indeed attributes like `DepthVis` are functions. 

### Settings

### How can the issue be reproduced?

1. clone AirSim with the specified commit above
2. run `hello_car.py`.

### Include full error message in text form
The full output is below:
```
$ python hello_car.py 
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

API Control enabled: True
Saving images to /tmp/airsim_car
Speed 0, Gear 1
Go Forward
Go Forward, steer right
Go reverse, steer right
Apply brakes
Traceback (most recent call last):
  File "hello_car.py", line 66, in <module>
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGB array
  File "/xxxAirSim/PythonClient/airsim/client.py", line 264, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name)
  File "/xxx/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/xxx/python3.7/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/xxx/python3.7/site-packages/msgpackrpc/transport/tcp.py", line 94, in send_message
    sock.send_message(message, callback)
  File "/xxx/python3.7/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "msgpack/_packer.pyx", line 284, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 290, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 287, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 234, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 277, in msgpack._packer.Packer._pack
  File "/xxx/python3.7/site-packages/msgpackrpc/transport/tcp.py", line 12, in <lambda>
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
AttributeError: 'function' object has no attribute 'to_msgpack'
```

**What's better than filing an issue? Filing a pull request :).**

zkytony, 2021-04-01:
Turns out the following worked
```python
    responses = client.simGetImages([
        airsim.ImageRequest("0", 3),
        airsim.ImageRequest("1", 2),
        airsim.ImageRequest("1", 0),
        airsim.ImageRequest("1", 0, False, False)])
```
Basically replacing e.g. `airsim.ImageType.DepthVis` by its corresponding int.

mmorphew, 2021-04-01:
This also fixes an identical error in hello_drone.py, discussed in #3533. Thank you for finding this fix! It appears that the metaclass `_ImageType` was recently added to address a renaming issue with `DepthPlanner`, however this inadvertently broke something in the image request pipeline. @saihv Just wanted to give you a heads-up about this error, as it does seem to stem from your last commit to `types.py`.

zkytony, 2021-04-02:
I wonder why I don't see the three camera images at the bottom of the screen (depth, RGB and semantic) when I hit "Play". Could it be that this issue affects that too?

ahmed-elsaharti, 2021-04-03:
I think I ran into this issue a couple of months ago and fixed it by uninstalling `airsim` and `msgpack-rpc-python` using 
```
pip uninstall airsim
pip uninstall msgpack-rpc-python
```
then reinstalling them using
```
pip install airsim
pip install msgpack-rpc-python
```
Could you guys try this and verify if its a valid solution?

Roopesh-Bharatwaj-K-R, 2021-04-04:
@ahmed-elsaharti , I tried to edit the Code in the Hello drone and changed. 

    responses = client.simGetImages([
        airsim.ImageRequest("0", 3),
        airsim.ImageRequest("1", 2),
        airsim.ImageRequest("1", 0),
        airsim.ImageRequest("1", 0, False, False)])
Basically replacing e.g. airsim.ImageType.DepthVis by its corresponding int.

And i  commented out the _ImageType class and the second ImageType class in PythonClient/airsim/ types.py. 

And i Encounter a different error for the image :  List Object has no Attribute - 'pixel-as-float'
[image]

zkytony, 2021-04-05:
> I think I ran into this issue a couple of months ago and fixed it by uninstalling `airsim` and `msgpack-rpc-python` using
> 
> ```
> pip uninstall airsim
> pip uninstall msgpack-rpc-python
> ```
> 
> then reinstalling them using
> 
> ```
> pip install airsim
> pip install msgpack-rpc-python
> ```
> 
> Could you guys try this and verify if its a valid solution?
> 
> Edit:
> actually, try using `pip list` to see if you have any other msgpack modules installed, uninstall those then reinstall msgpack-rpc-python and airsim. in my case I also uninstalled `msgpack-python` and `msgpack`

I didn't need to reinstall those two packages. But indeed `msgpack-rpc-python` is the one to use, according to the [documentation](https://microsoft.github.io/AirSim/apis/).

zkytony, 2021-04-06:
> @ahmed-elsaharti , I tried to edit the Code in the Hello drone and changed.
>
> ```
> responses = client.simGetImages([
>     airsim.ImageRequest("0", 3),
>     airsim.ImageRequest("1", 2), [...]

I'm not sure about that. What's the object type of `response`? Does it have `pixels_as_float` as an attribute? Does it match the most recent version on this repository?

Roopesh-Bharatwaj-K-R, 2021-04-06:
Hi @zkytony , @ahmed-elsaharti   Have Tried couple of times this one but still we face same problem. Thanks

yuzhijia1997, 2021-04-13:
@Roopesh-Bharatwaj-K-R ,Hello,have you solved this problem,we also had the same problem

Roopesh-Bharatwaj-K-R, 2021-04-14:
> [CAUTION]: This email originated from an external source. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi, i have solved the problem..and i used the old Airsim package ...new airsim which was cloned was making us trouble.

Tá an t-eolas atá le fáil sa ríomhphost seo faoi iontaoibh agus tá sé ceaptha le haghaidh aird an fhaighteora bheartaithe/na bhfaighteoirí beartaithe amháin. Más rud é go bhfuair tú an ríomhphost seo go hearráideach, ná húsáid agus ná tarchuir é ar mhaithe le haon chuspóir, le do thoil; ina áit sin cuir ar an eolas muid láithreach agus scrios gach cóip den ríomhphost seo ó do chóra(i)s ríomhaireachta. Ach amháin sa chás gur comhaontaíodh a leithéid go sonrach ag ár n-ionadaí údaraithe, is le húdar an ríomhphoist amháin na tuairimí a chuirtear in iúl ann, agus ní léiríonn siad tuairim ná ní chuireann siad ceangal ar aon chaoi eile ar Institiúid Teicneolaíochta Bhaile Átha Luain. Déan teagmháil le ***@***.*** nó cuir glao ar 090 6468000. The information contained in this email is confidential and is designated solely for the attention of the intended recipient(s). If you have received this email in error, please do not use or transmit it for any purpose but rather notify us immediately and delete all copies of this email from your computer system(s). Unless otherwise specifically agreed by our authorised representative, the views expressed in this email are those of the author only and shall not represent the view of or otherwise bind Athlone Institute of Technology. Contact ***@***.*** or telephone 090 6468000.

ahmed-elsaharti, 2021-04-14:
@Roopesh-Bharatwaj-K-R  Based on [your reply](https://github.com/microsoft/AirSim/issues/3550#issuecomment-815919462) on #3550 it looks like you also moved to a new machine. So I wouldn't conclude that the Airsim version was the issue since it was tested on a completely different machine (where the other machine probably has a cleaner install of python altogether (which probably is the actual culprit behind the error))

Roopesh-Bharatwaj-K-R, 2021-04-14:
@ahmed-elsaharti , No we did not move to new Machine but we had the same old package of Airsim on another , which we copied and run on the same old machine which worked well.

ahmed-elsaharti, 2021-04-14:
@jonyMarino this seems like a bug worth tracking down.

PR #3594 (2021-04-15): remove duplicate definition of ImageType class

Fixes: #3533
Fixes: #3553
Related to: #3550 

## About
This PR addresses an error where accessing fields in the ImageType class would generate an error: "AttributeError: 'function' object has no attribute 'to_msgpack'". The fix is to remove a duplicate definition of the ImageType class in types.py

## How Has This Been Tested?
hello_drone.py script was run and confirmed to complete execution without error

## Screenshots (if appropriate):
closed 2021-04-15 by jonyMarino