microsoft/AirSim #3557: About the distance measurement

git-liweichao, 2021-04-01:
## Question
### What's your question?

Hi, I have a question about the Airsim simulation platform.  I want to measure the distance between the quadcopter and the target on the ground. I think it maybe a slash. I don't know whether the Lidar sensor in airsim can achieve that. If it can, how can I setting? Thanks for you help! 
### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

ahmed-elsaharti, 2021-04-01:
Hi @git-liweichao  Are you trying to get the 'ground truth' distance between the drone and the object or do you want to obtain sensor-based calculation of the distance between the drone and the object?

git-liweichao, 2021-04-06:
> Hi @git-liweichao Are you trying to get the 'ground truth' distance between the drone and the object or do you want to obtain sensor-based calculation of the distance between the drone and the object?

Thanks for your reply. I want to obtain sensor-based calculation of the distance between the drone and the object. If I can also get the 'ground truth' distance between the drone and the object, it will be more helpful for me.

jonyMarino, 2021-04-16:
Hi and welcome @git-liweichao! you can accomplish that using `simGetObjectPose` and `` and subtracting the positions. Take a look at https://microsoft.github.io/AirSim/apis/#common-apis and [simGetVehiclePose](https://microsoft.github.io/AirSim/api_docs/html/index.html?highlight=simgetvehiclepose#airsim.client.VehicleClient.simGetVehiclePose )

closed 2021-04-16 by jonyMarino