microsoft/AirSim #3559: Building of Hello Drone

Roopesh-Bharatwaj-K-R, 2021-04-01:
## Question
### where to code after the Hello Drone running, for RL implementation and Object Detection 

### Include context on what you are trying to achieve

### Include details of what you already did to find answers

ahmed-elsaharti, 2021-04-02:
>where to code after the Hello Drone running, for RL implementation and Object Detection

Similar to #3550, If you're planning on using the [python APIs](https://microsoft.github.io/AirSim/apis/), all you would need to do is make sure the airsim package is installed through pip and you can start using the APIs similar to the [demo python scripts](https://github.com/microsoft/AirSim/tree/master/PythonClient/multirotor).

>How i can make the drone movements autonomous 

you can use the [moveToPositionAsync](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/PythonClient/airsim/client.py#L982), the [moveByVelocityAsync](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/PythonClient/airsim/client.py#L959) or any of the other movement API calls. For examples on how to use these, check the [demo files](https://github.com/microsoft/AirSim/tree/master/PythonClient/multirotor)

>which file do i need to code for the object detection Algorithm

You can create your own python script as long as you `import airsim`. Or you can modify any of the [demo python scripts](https://github.com/microsoft/AirSim/tree/master/PythonClient/multirotor)

Roopesh-Bharatwaj-K-R, 2021-04-02:
Thank You very Much, really Helpful !

closed 2021-04-16 by jonyMarino