microsoft/AirSim #3560: moveToPositionAsync in Flask route

seandiacono, 2021-04-02:
## Question
### What's your question?

Why does moveToPositionAsync not work in a Flask route?
### Include context on what you are trying to achieve

I'm creating a Flask API to control a drone in AirSim from a Web application. So an API call is sent to Flask which then sends commands to AirSim running on the same machine as the Flask server. However the moveToPositionAsync function does not work in route whilst the takeoffAsync and landAsync functions do work. I'm not sure why this would be the case since all three functions return the same type of Future.
#### Context details
```python
@app.route('/sendFlightParams', methods=['POST'])
@cross_origin()
def set_flight_params():
    data = request.get_json()

    x = data['xCoord']
    y = data['yCoord']
    altitude = data['altitude']
    velocity = data['velocity']

    client = airsim.MultirotorClient()
    client.confirmConnection()
    client.enableApiControl(True)

    client.takeoffAsync().join() # this works
    print("Takeoff Complete")
    client.moveToPositionAsync(0, 0, altitude, 2).join() # this doesn't work
    client.landAsync().join() # this works

    status = {"status": "OK"}

    return status
```

### Include details of what you already did to find answers

I tried to remove the join()s however it didn't help and I tried using asyncio but it didn't help either. I also asked the question on [Stack Overflow](https://stackoverflow.com/questions/66917587/flask-route-not-waiting-for-future) but I feel like this question is more appropriate here.

ahmed-elsaharti, 2021-04-02:
I just tested this with 
```python
@app.route('/sendFlightParams.php')
@cross_origin()
def set_flight_params():

    client = airsim.MultirotorClient()
    client.confirmConnection()
    client.enableApiControl(True)

    client.takeoffAsync().join()
    print("Takeoff Complete")
    client.moveToPositionAsync(0, 0, -30, 2).join() 
    client.landAsync().join()

    status = {"status": "OK"}

    return status
```
and it worked fine. 

Based on your [Stack Overflow post](https://stackoverflow.com/questions/66917587/flask-route-not-waiting-for-future) it looks like you're using the Z value from `data['altitude']` which could be set to a Z value very similar to the drone's altitude after takeoff resulting in the `client.moveToPositionAsync(0, 0, altitude, 2).join()` call being skipped (I'm assuming it's being skipped since you said landing gets executed)

jonyMarino, 2021-04-16:
Hi and welcome @seandiacono! Have you tried with Ahmed's code and are still having problems??

seandiacono, 2021-04-30:
@jonyMarino Yes, I tried the code and I am still having issues with it, I moved to the FastApi framework and the issue was solved.

closed 2021-05-02 by seandiacono