microsoft/AirSim #3567: Gyroscope error after reset airsim

kwanyongcho7, 2021-04-06:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version: 
- autopilot version: Px4
- OS Version: Windows 10

I used Airsim Reset. I reset my drone when it is upside down, I can see the drone correctly but QGC's gyroscope(next to the compass) kept upside down which makes me cannot takeoff. Is there any way I can reset gyroscope sensor? 

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2021-04-06:
If I'm understanding the issue correctly, [this comment](https://github.com/microsoft/AirSim/discussions/3474#discussioncomment-494125)  might offer an explanation as to why the sensor is acting this way

jonyMarino, 2021-04-16:
Hi @kwanyongcho7! That's correct, Ahmed; currently, PX4 doesn't support a software reset. So you need to re-run PX4 and reset AirSim to accomplish that.

kwanyongcho7, 2021-04-20:
Can I just stay the current state of Drone in Airsim? For now, Airsim just resets(fix upside down) when my drone is rolling.. I want to just keep the reversed shape.

jonyMarino, 2021-04-22:
I'm not sure where that happens. Someone needs to trace this to find the root of that.

jonyMarino, 2021-04-22:
you can accomplish that by setting the drone's orientation with setVehiclePose() after the reset.

closed 2021-04-22 by jonyMarino