microsoft/AirSim #3569: some uasset in AirSim1.4.0 can't support UE4.24

xyxu1024, 2021-04-07:
## Bug report 

- AirSim Version/#commit: 1.4.0
- UE/Unity version: UE4.24.
- autopilot version: None
- OS Version: Windows

### What's the issue you encountered?

I tried to introduce AirSim-1.4.0 (latest at present) my to previous UE project, and the UE project must run under UE4.24. 
I looked up  [AirSim Document](https://microsoft.github.io/AirSim/build_windows/) and it said that
> Note: AirSim also works with UE >= 4.24, however, we recommend 4.25.

This AirSim-1.4.0 plugin is built and compiled in UE4.24 as "Develop Editor" successfully under the AirSim document.
But when I run it, the UE4.24 editor said:
[image]
These warings and error can be ignored to open the editor, but it can't run in editor because some exceptions:
[image]

I looked up the debug log, it said:
```
[2021.04.07-02.39.28:279][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
[2021.04.07-02.39.28:280][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
[2021.04.07-02.39.28:280][  0]LogUObjectGlobals: Warning: find object “Class /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C” failed
CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
[2021.04.07-02.39.28:280][  0]Error: CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:287][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
CDO Constructor (ComputerVisionPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
[2021.04.07-02.39.28:288][  0]Error: CDO Constructor (ComputerVisionPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:306][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/HUDAssets/CameraDistortion.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
[2021.04.07-02.39.28:306][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/HUDAssets/CameraDistortion.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
[2021.04.07-02.39.28:306][  0]LogUObjectGlobals: Warning: find object “Material /AirSim/HUDAssets/CameraDistortion.CameraDistortion” failed
CDO Constructor (PIPCamera): Failed to find Material'/AirSim/HUDAssets/CameraDistortion.CameraDistortion'
[2021.04.07-02.39.28:306][  0]Error: CDO Constructor (PIPCamera): Failed to find Material'/AirSim/HUDAssets/CameraDistortion.CameraDistortion'

[2021.04.07-02.39.28:342][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
[2021.04.07-02.39.28:343][  0]Error: CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:393][  0]LogLinker: Warning: Asset 'E:/AirSim-1.4.0/Unreal/Environments/Blocks/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' has been saved with engine version newer than current and therefore can't be loaded. CurrEngineVersion: 4.24.3-11590370+++UE4+Release-4.24 AssetEngineVersion: 4.25.0-13144385+++UE4+Release-4.25
CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
[2021.04.07-02.39.28:393][  0]Error: CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: -------------- Default Property warnings and errors:
[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: Error: CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: Error: CDO Constructor (ComputerVisionPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: Error: CDO Constructor (PIPCamera): Failed to find Material'/AirSim/HUDAssets/CameraDistortion.CameraDistortion'

[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: Error: CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

[2021.04.07-02.39.28:452][  0]LogUObjectBase: Warning: Error: CDO Constructor (SimModeBase): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C
```
It seems like the uasset files <code>Blueprints/BP_PIPCamera.uasset</code>, <code>HUDAssets/CameraDistortion.uasset</code> are incompatible with UE4.24, and I finded that [AirSim Changelog](https://microsoft.github.io/AirSim/CHANGELOG/) said that AirSim-1.4.0 added some new features about cameras and their distortion in Dec 2020.

**I was just wondering HOW I could modify these ussat files to make them forward compatible. They are seemly binary files.**

### Settings

Using default 
### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-04-12:
Hi @xyxu1024! And welcome! The documentation reflects the state of the master branch. Version 1.4 supports UE >=4.25.
If you need to use 4.24, please clone the repo with the latest commit. Sorry for the incumbencies.

xyxu1024, 2021-04-13:
Thanks @jonyMarino! I tried it, but uasset in the latest repo was generated by ue 4.25: [BP_PIPCamera.uasset](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset), [CameraDistortion.uasset](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Content/HUDAssets/CameraDistortion.uasset). These files are raw and I guess they were build by their contributor using ue 4.25 @saihv

jonyMarino, 2021-04-13:
I thought PR #3404 fixed that...

jonyMarino, 2021-05-06:
It is not a simple task for us to recompile those assets for 4.24. So, we are going to support >=4.25 as we were stating before #3404.

closed 2021-06-15 by jonyMarino