microsoft/AirSim #3582: error initializing SDL

shanek16, 2021-04-12:
## Bug report 

- AirSim Version/#commit: https://github.com/microsoft/AirSim/commit/d3e4185d615acbe93b1c46c55f90e53b5e1061a5
- UE/Unity version: https://github.com/EpicGames/UnrealEngine/commit/5df54b7ef1714f28fb5da319c3e83d96f0bedf08
- autopilot version: N/A
- OS Version: Ubuntu 18.04.3 LTS

### What's the issue you encountered?

Issue occured while following https://microsoft.github.io/AirSim/build_linux/
same issue occurs while ~/Downloads/Blocks/LinuxNoEditor$ ./Blocks.sh 
after downloading binaries from https://github.com/Microsoft/AirSim/releases

It seems like something with GPU driver is related, but not sure which commands to execute
Thank you for your help :)

### Settings
](url)

result of nvidia-smi:
Mon Apr 12 11:35:45 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.87.01    Driver Version: 418.87.01    CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN Xp            On   | 00000000:02:00.0 Off |                  N/A |
| 23%   29C    P8     9W / 250W |    179MiB / 12194MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  TITAN Xp            On   | 00000000:03:00.0 Off |                  N/A |
| 23%   29C    P8     9W / 250W |     12MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   2  TITAN Xp            On   | 00000000:82:00.0 Off |                  N/A |
| 23%   30C    P8     8W / 250W |     12MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   3  TITAN Xp            On   | 00000000:83:00.0 Off |                  N/A |
| 23%   33C    P8     9W / 250W |     12MiB / 12196MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1845      G   /usr/lib/xorg/Xorg                             9MiB |
|    0      1959      G   /usr/bin/gnome-shell                          10MiB |
|    0     19681      C   /usr/lib/libreoffice/program/soffice.bin     147MiB |
+-----------------------------------------------------------------------------+

### How can the issue be reproduced?

1. follow steps(Build on Linux) from:
https://microsoft.github.io/AirSim/build_linux/#host-machine
2.Go to UnrealEngine installation folder and start Unreal by running ./Engine/Binaries/Linux/UE4Editor.

### Include full error message in text form
~/Project/UnrealEngine$ ./Engine/Binaries/Linux/UE4Editor
Increasing per-process limit of core file size to infinity.
LogInit: Display: Running engine without a game
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 53 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.101775
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +9:00, Platform Override: ''
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin ActorLayerUtilities
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
[... 206 lines cut ...]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.AllowParallelGC:1]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.TimeBetweenPurgingPendingKillObjects:61.1]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.MaxObjectsInEditor:25165824]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.IncrementalBeginDestroyEnabled:1]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.CreateGCClusters:1]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.MinGCClusterSize:5]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.ActorClusteringEnabled:0]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.BlueprintClusteringEnabled:0]]
[2021.04.12-01.30.48:253][  0]LogConfig: Setting CVar [[gc.UseDisregardForGCOnDedicatedServers:0]]
[2021.04.12-01.30.48:254][  0]LogConfig: Applying CVar settings from Section [/Script/Engine.NetworkSettings] File [../../../Engine/Saved/Config/Linux/Engine.ini]
[2021.04.12-01.30.48:254][  0]LogConfig: Applying CVar settings from Section [/Script/UnrealEd.CookerSettings] File [../../../Engine/Saved/Config/Linux/Engine.ini]
[2021.04.12-01.30.48:254][  0]LogInit: Initializing SDL.
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  98
  Current serial number in output stream:  99
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Segmentation fault (core dumped)

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-04-21:
Hi and welcome @shanek16! Can you try with the docker option? Also, take a look at the steps in #2991

zimmy87, 2022-01-14:
Hi @shanek16, are you still experiencing this issue?