microsoft/AirSim #3585: ExponentialHeightFog

icisneros, 2021-04-12:
Is there a way to prevent Airsim from spawning this element?  Currently, I have to manually reduce the Fog level to 0 every time I start a new session, but this is cumbersome and prevents automating my workflow.  The python script that I run does not make any calls to simEnableWeather, so I don't know why Airsim spawns this element at all.

[image]

zimmy87, 2021-05-07:
Hi @icisneros, thanks for opening this issue. The ExponentialHeightFog actor is spawned inside the BeginPlay event of Blueprint'/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor'. You can either prevent this actor from spawning by disconnecting the call to SpawnActor there or you can apply #3666 which will set this actor to be invisible until a weather parameter is set.

closed 2021-05-12 by zimmy87

icisneros, 2021-05-17:
Thanks for the response!  

I am somewhat new to this, and I haven't previously made changes to plugin files.  I have copied your versions of WeatherLib.cpp and WeatherLib.h from #3666 into my "MyCustomProject/Plugins/Airsim/Source/Weather/" directory, but now I am not sure how to compile this so that the changes take effect.  Should I have instead copied them into my local repo clone in "Airsim/Unreal/Plugins/Airsim/Source/Weather/" and run ./setup.sh and ./build.sh ?  I am working in Ubuntu.

PR #3666 (2021-05-12): add logic for setting visibility of ExponentialHeightFog actor

Fixes: #3585    

## About
set visibility of the ExponentialHeightFog actor to false when it is spawned inside WeatherActor and then update the visibility to true once a weather parameter has been altered

## How Has This Been Tested?
weather parameters were set via the F10 weather menu as well as calling the client.simEnableWeather() and client.simSetWeatherParameter() APIs from python

## Screenshots (if appropriate):