microsoft/AirSim #3592: Moving Co-ordinates in Drone

Roopesh-Bharatwaj-K-R, 2021-04-14:
How do we need to set the Position Co-ordinates for the Drone to reach the Target location and in which file i need to modify code For this.  Can we Use PX4 Autopilot Software, 

i Tried running a sample DQN Code on drones it Move toward the sky and then it resets to the original position ... In this Scenario , i want to set a Location on the environment and how will i fetch the co-ordinates of the point created and where will i have to update the same in the Airsim to travel to that point.

Roopesh-Bharatwaj-K-R, 2021-04-14:
Hi, @ahmed-elsaharti any suggestions for this.

ahmed-elsaharti, 2021-04-16:
Hi @Roopesh-Bharatwaj-K-R, 
To set a particular way-point for the drone to fly to, you could use the [moveToPositionAsync API](https://github.com/microsoft/AirSim/blob/210f3eee28169b07bf9e3e3b59109b047ed4ab1a/PythonClient/airsim/client.py#L991) similar to these examples:
[1](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/PythonClient/multirotor/gimbal.py#L18), [2](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/PythonClient/multirotor/survey.py#L37) and [3](https://github.com/microsoft/AirSim/blob/6d20663e0818448613be0332009b02bd6341589e/PythonClient/multirotor/reset_test_drone.py#L13). And this can also be done with the same API when running in PX4 mode.

In reference to the question about where to code:
Similar to [this comment](https://github.com/microsoft/AirSim/issues/3559#issuecomment-812732852):
>You can create your own python script as long as you import airsim. Or you can modify any of the demo python scripts

Roopesh-Bharatwaj-K-R, 2021-04-18:
@ahmed-elsaharti Hi , Thank you very much

closed 2021-04-21 by jonyMarino