microsoft/AirSim #3600: I cannot run "set_wind.py"

4iue0, 2021-04-15:
## I cannot run "set_wind.py"
### Please tell me how to run set_wind.py

### I've been running AirSim for other things, but I needed to simulate wind, and when I tried to run the tutorial set_wind.py, I couldn't run it.

#### 

OS : windows 10 pro
AirSim version: 1.3.1(maybe... I wanna know AirSim version) 
Binary Blocks

I got the following error.
``Traceback (most recent call last):
  File "set_wind.py", line 13, in <module>
    client.simSetWind(wind)
  File "C:\AirSim\PythonClient\airsim\client.py", line 791, in simSetWind
    self.client.call('simSetWind', wind)
  File "C:\Users\USER\anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\USER\anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simSetWind' with argument count 1.``

Please tell me.

rajat2004, 2021-04-15:
The version you're using is old, try it with 1.4.0. The [Releases page](https://github.com/microsoft/AirSim/releases) has the assets for each version

4iue0, 2021-04-16:
@rajet2004
Thank you for replying.
I will try it!

closed 2021-05-25 by 4iue0