microsoft/AirSim #3602: ./build.sh

EdwinMeriaux, 2021-04-16:
hey!

I recently modified the mass of a drone in the Multirotorparams.hpp  to 1.0f.
Every time the moment I do that when ever I run: ./build.sh I get:

+ debug=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ folder_name=
+ [[ ! -d build_release ]]
+ pushd build_release
++ nproc
+ make -j64
[ 41%] Built target MavLinkCom
[ 47%] Built target MavLinkTest
[ 67%] Built target rpc
Scanning dependencies of target AirLib
make[2]: *** No rule to make target '/home/edwinm/AirSim/AirLib/src/api/RpcLibClientBase.cpp', needed by 'AirLib/CMakeFiles/AirLib.dir/home/edwinm/AirSim/AirLib/src/api/RpcLibClientBase.cpp.o'.  Stop.
CMakeFiles/Makefile2:159: recipe for target 'AirLib/CMakeFiles/AirLib.dir/all' failed
make[1]: *** [AirLib/CMakeFiles/AirLib.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

would anyone know why?

EdwinMeriaux, 2021-04-16:
If I first run ./clean.sh then run ./build.sh I get:

+ debug=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ folder_name=
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Running CommonSetup...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- found AIRSIM_ROOT=/home/edwinm/AirSim
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Success
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Success
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
You have called ADD_LIBRARY for library AirLib without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/edwinm/AirSim
[... 554 lines cut ...]
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x278): undefined reference to `msr::airlib::MultirotorApiBase::setVelocityControllerGains(std::__1::vector<float, std::__1::allocator<float> > const&, std::__1::vector<float, std::__1::allocator<float> > const&, std::__1::vector<float, std::__1::allocator<float> > const&)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x280): undefined reference to `msr::airlib::MultirotorApiBase::setPositionControllerGains(std::__1::vector<float, std::__1::allocator<float> > const&, std::__1::vector<float, std::__1::allocator<float> > const&, std::__1::vector<float, std::__1::allocator<float> > const&)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x288): undefined reference to `msr::airlib::MultirotorApiBase::setSafetyEval(std::__1::shared_ptr<msr::airlib::SafetyEval>)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x290): undefined reference to `msr::airlib::MultirotorApiBase::setSafety(common_utils::EnumFlags<msr::airlib::SafetyEval::SafetyViolationType_, unsigned int>, float, msr::airlib::SafetyEval::ObsAvoidanceStrategy, float, Eigen::Matrix<float, 3, 1, 0, 3, 1> const&, float, float, float)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x298): undefined reference to `msr::airlib::MultirotorApiBase::moveByRollPitchYawZInternal(float, float, float, float)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x2a0): undefined reference to `msr::airlib::MultirotorApiBase::moveByRollPitchYawThrottleInternal(float, float, float, float)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x2a8): undefined reference to `msr::airlib::MultirotorApiBase::moveByRollPitchYawrateThrottleInternal(float, float, float, float)'
CMakeFiles/AirLibUnitTests.dir/home/edwinm/AirSim/AirLibUnitTests/main.cpp.o:(.data.rel.ro._ZTVN3msr6airlib15SimpleFlightApiE[_ZTVN3msr6airlib15SimpleFlightApiE]+0x2b0): undefined reference to `msr::airlib::MultirotorApiBase::moveByRollPitchYawrateZInternal(float, float, float, float)'
[... cut ...] piBase'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/build.make:97: recipe for target 'output/bin/AirLibUnitTests' failed
make[2]: *** [output/bin/AirLibUnitTests] Error 1
CMakeFiles/Makefile2:328: recipe for target 'AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all' failed
make[1]: *** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

rajat2004, 2021-04-16:
That's pretty strange, just tested the latest master and working fine. What's the exact commit you're working on? I wonder if you somehow are in the middle of a series of commits from a PR

EdwinMeriaux, 2021-04-16:
@rajat2004 I am very  confused I even reinstalled Airsim and I get the same issue. 
git clone https://github.com/Microsoft/AirSim.git

Was that not the proper install?

xxEoD2242, 2021-04-16:
@EdwinMeriaux It appears that running the clean command is not actually removing the binaries that you compiled from a previous version. Did you change anything besides AirLib?

Usually, when I have this problem, it's related to the CMake cache and how the system pulls binaries to link. This will remove your work but you could try saving the work that you did (i.e. the files) in another directory and then run `git clean -dfx` or `git reset --hard` to revert back to the pre-installed version, then apply your changes. Either of these commands should remove the binaries from the CMake Cache.

rajat2004, 2021-04-16:
`clean_rebuild.sh` will do the same as pointed out by @xxEoD2242, but yeah, save any specific work by copying the files or `git stash`

EdwinMeriaux, 2021-04-17:
Ok so if I understand correctly I have not properly removed all the files from a previous install of Airsim so I need to clean first and then rebuild using the clean__rebuild.sh?

rajat2004, 2021-04-17:
Just clean_rebuild.sh should be enough, but save any work required

EdwinMeriaux, 2021-04-21:
Sweet! yeah that worked!

closed 2021-04-21 by EdwinMeriaux

blackcyavuz, 2023-04-23:
> Just clean_rebuild.sh should be enough, but save any work required Most likely the case of old files, otherwise much more of a headache to figure out why this is happening

it worked thanks