microsoft/AirSim #3613: No compatible vehicles for SimMode in settings.json when starting AirSim with Python in the Unreal Editor

LeaRain, 2021-04-20:
## Bug report 

- AirSim Version/#commit: v1.4.0 Windows
- UE/Unity version: UE 4.26.1
- autopilot version: NA
- OS Version: Windows 10

### What's the issue you encountered?

I am trying to automate most of my work with AirSim and Unreal Engine, so I want to start AirSim/my Unreal Engine game in the editor with the command `unreal.EditorLevelLibrary.editor_play_simulate()`. When I press the Play button, everything is working fine, but with the command I get the error `There were no compatible vehicles created for current SimMode! Check your settings.json.`  I tested it with a custom environment I found in the Unreal Marketplace (City Park Environment Collection if that matters) and with the Blocks environment. I also tried to change the SimMode from ComputerVision (my default mode) to Multirotor, but still the same error message.

### Settings

My settings.json contains:
```
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "CameraDefaults": {
        "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 1920,
                "Height": 1080,
                "FOV_Degrees": 90,
                "AutoExposureSpeed": 100,
                "MotionBlurAmount": 0
            },
            {
                "ImageType": 5,
                "Width": 1920,
                "Height": 1080,
                "FOV_Degrees": 90,
                "AutoExposureSpeed": 100,
                "MotionBlurAmount": 0
            }
        ]
    },
    "Cameras": [
        {
            "CameraName": "3",
            "ImageType": 0
        }
    ],
    "SimMode": "ComputerVision",
    "ViewMode":  "NoDisplay"
}
```

### How can the issue be reproduced?

1. Open the .sln file of the environment in Visual Studio (2019) and run the project
2. Install the plugins "Python Editor Script Plugin" and "Sequencer Scripting", if it's not already installed
3. Run the command `unreal.EditorLevelLibrary.editor_play_simulate()` in a Python command line and hopefully reproduce the bug

### Include full error message in text form
There were no compatible vehicles created for current SimMode! Check your settings.json.

Thank you for your help and I'm sorry if I just misse a setting for solving this bug or so.

saihv, 2021-04-20:
@LeaRain The 'SImulate' option in UE starts the game and just lets it run, without taking possession of the main pawn (vehicle), which is very likely causing the issue where AirSim does not find a vehicle to control. The Editor GUI has two options: Play (Alt+P) or Simulate (Alt+S). Is there an equivalent Python command to just play the game instead of simulate?

LeaRain, 2021-04-21:
@saihv My apologies, I wasn't aware of the difference between simulate and actually play. I don't know if there is an equivalent Python command just for the play mode, at least, I couldn't find any. I'll guess I just ask in the Unreal Engine forum. Thank you for your help! :)

closed 2021-04-21 by LeaRain

Nono0118, 2024-01-18:
Hello, I am trying to run airsim in a newly created ue environment, but the same problem as you appears, have you solved this problem and how to solve it? Thank you for your help.