microsoft/AirSim #3614: rotateToYawAsync() does nothing

MostafaAhmed95, 2021-04-20:
## Bug report 

- AirSim Version/#commit: v1.2.1
- UE/Unity version:
- autopilot version:
- OS Version: Windows

### What's the issue you encountered?

I downloaded Mountain Landscape (with power lines) environment v1.2.1 for windows and I use this simple code with it, but my drone doesn't change it's yaw angle at all 
`import airsim
import os

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

# Async methods returns Future. Call join() to wait for task to complete.
client.takeoffAsync().join()
client.rotateToYawAsync(60).join()
`
### Settings

https://gist.github.com/MostafaAhmed95/12d978355cdd3f81bc17f52b6fdc3bcf

### How can the issue be reproduced?

1. download [Mountain Landscape (with power lines)](https://github.com/microsoft/AirSim/releases/tag/v1.2.1) on windows
2. Run it using .\run.bat
3. in another power shell run the python code mentioned above

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2021-04-21:
Hi @MostafaAhmed95, I was able to reproduce the issue (only when using the same settings for some reason) which leads me to believe that this could have been a bug in the old version of AirSim (1.2.1). Try using the [latest release](https://github.com/microsoft/AirSim/releases/tag/v1.4.0-windows)

MostafaAhmed95, 2021-04-21:
Thanks @ahmed-elsaharti for your try, I tried the same environment in the latest release and also the blocks environment and both worked fine, so as you said it seems to be a bug, But the idea here is that the latest version of Landscape Mountains environment doesn't have power lines models which I need to work on them, so do you have any idea how I can work around this to get the old version of this environment work properly maybe change something in the setting or what files should I make changes to in the code base

ahmed-elsaharti, 2021-04-22:
@MostafaAhmed95, Surprisingly, the simulation seemed to perform well on version 1.2.1 and the `rotateToYawAsync` call worked perfectly with the default settings. This leads me to assume that it's a combination of both the settings you're using and the old version that is causing this issue to occur.
You'll have to either mess around with the settings.json file to figure out what setting is causing this to happen OR you could [Create a custom environment](https://microsoft.github.io/AirSim/unreal_custenv/) with the Landscape mountains level and add transmission lines manually ([the docs](https://microsoft.github.io/AirSim/unreal_custenv/) and [this youtube tutorial](https://www.youtube.com/watch?v=y09VbdQWvQY) might help)

MostafaAhmed95, 2021-04-23:
@ahmed-elsaharti, if you are able to share with me your settings.json file so I could experience with it, it will be great, Thanks for suggestion to build a custom environment, I will try it as a last resort.

ahmed-elsaharti, 2021-04-30:
@MostafaAhmed95 here are the default settings:
```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2
}

```

closed 2022-05-30 by jonyMarino