microsoft/AirSim #3623: simGetImages() returns an empty image (i.e. empty image_data_uint8 array)

InfraredLaser, 2021-04-23:
## Bug report 

- AirSim Version/#commit: 1.3.0
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

Sometimes simGetImages() returns a image_data_uint8 array of size 0.  This makes it impossible to resize or view an image captured by the AirSim API. 
 
### Settings

{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "EngineSound": false,

  "PawnPaths": {
    "DefaultCar": { "PawnBP": "Class'/Game/Blender/RA/AGV3/BP_AGV_3.BP_AGV_3_C'"},
    "DefaultQuadrotor": {"PawnBP": "Class'/AirSim/Blueprints/BP_FlyingPawn.BP_FlyingPawn_C'"}
  },

 
  "DefaultSensors": {
    "MyLidar1": {
      "SensorType": 6,
      "Enabled": false,
      "NumberOfChannels": 8,
      "PointsPerSecond": 500,
      "X": 0,
      "Y": 0,
      "Z": -1,
      "DrawDebugPoints": false
    },
    "DistanceSensor":{
      "SensorType": 5,
      "Enabled": true,
      "MaxDistance": 40,
      "MinDistance": 1,
      "DrawDebugPoints": false
    }
  
    
  },

  "CameraDefaults": {
    "CaptureSettings":[
      { 
        "ImageType": 0,
        "Width": 256,
        "Height": 144,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }

}
### How can the issue be reproduced?

1. Running the image capture in a loop to display the images is the best chance to reproduce the error.
2. Here is the code I am running to change the image taken from simGetImages into a 2d image:

```
images = self.client.simGetImages([airsim.ImageRequest("0", airsim.ImageType.Scene, False, False), airsim.ImageRequest("0", airsim.ImageType.DepthPerspective, pixels_as_float=True)]) 

scene_1d = np.frombuffer(images[0].image_data_uint8, dtype=np.uint8)
scene_2d = scene_1d.reshape(IMG_HEIGHT, IMG_WIDTH, 3)
```

3. IMG_HEIGHT and IMG_WIDTH are taken from the settings.json file under "CaptureSettings".

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-04-23:
Hi @InfraredLaser! That is an old issue maybe was present in v1.3.0. Can you use v1.4.0?

InfraredLaser, 2021-04-23:
Hi @jonyMarino, pip installing airsim only installs v1.3.0. I can't seem to find v1.4.0. I have downloaded the latest windows .zip file of v1.4.0 and I could just copy and paste the python airsim folder to my python environment folder, but is there a cleaner way of doing so using pip?