microsoft/AirSim #3625: Problem moving the drone with MoveToPositionAsync

Agm2104, 2021-04-23:
## Bug report 

- AirSim Version/#commit: 1.3.1
- UE/Unity version:2.25.4

- autopilot version: PX4 1.10.1 SITL
- OS Version: Windows 10.0.19042.870

### What's the issue you encountered?

I'm trying to make a code for obstacle avoidance by using lidar, but whenever I try to send the drone to the next position, it keeps hovering over the same place without moving:

[image]

I get the RC and offboard error, which I try to solve by setting the NAV_RCL_ACT to 0 in the PX4 terminal, once the drone takes off, but it doesn't seem to work:
[image]

When time passes PX4 goes crazy and I get this:
[image]

As you can see, however, the code keeps compiling. it just seems unable to move the drone:
[image]

Any idea to solve this? Any answer would be much appreciated :)
Thanks in advance.

### Settings

[image]
### How can the issue be reproduced?

1.Download the code a and compile it while running UE and PX4
2.Check the python terminal and the UE screen. After setting the path, the drone should move to the first 7 positions one by one, but it stays in place. However, the code compiles properly. The code include several printed messages as 'milestones' to see what the dode should be doing.
3.You can use MoveToPositionAsync() before  the while loop and it works. It only fails inside the loop.

Here is the code:
https://gist.github.com/Agm2104/66d18b8c9276df4d6cada63a0b498b5e

### Include full error message in text form
This is what I get once a time passes and PX4 gets crazy:
pxh> WARN  [commander] Failsafe enabled: no RC and no offboard
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
ERROR [sensors] Mag #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Failsafe: stop and wait

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-04-28:
Hi @Agm2104! Can you try with the latest AirSim commit? We have fixed MoveToPositionAsync in #3545.
Also, you can control the AirSim drone using mavsdk instead of our API.

Agm2104, 2021-04-28:
Hi @jonyMarino, thanks for answering! How do I upgrade Airsim from 1.3.1 to 1.4? Do I need to build airsim from scratch or there is any other way?

jonyMarino, 2021-04-28:
AirSim 1.4 also has that bug. You need to clone this repository. Please follow the instruction under https://microsoft.github.io/AirSim/#how-to-get-it

lovettchris, 2021-07-08:
You might also get better results providing the full path in one call to moveOnPathAsync like this because it will smoothly connect each path segment that way otherwise the path segments will be disjoint and the drone will try and hover between each segment which can result in weird jittery flight:

moveOnPathAsync([airsim.Vector3r(125,0,z),
                                airsim.Vector3r(125,-130,z),
                                airsim.Vector3r(0,-130,z),
                                airsim.Vector3r(0,0,z)],
                        12, 120,
                        airsim.DrivetrainType.ForwardOnly, airsim.YawMode(False,0), 20, 1).join()

If you want to check the lidar and drone position while it is flying then remove the .join() so you can query the state of the drone while it's flying, you can save the "result" object and query whether it is complete in a loop.

jonyMarino, 2021-07-23:
Thanks Chris!

closed 2021-07-23 by jonyMarino