microsoft/AirSim #3637: Main_CityEnvironment.umap appears to be an asset file.

DanielAtt2000, 2021-04-29:
## Bug report 

- AirSim Version/#commit: 1.4.0
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I am currently trying to use and open the CityEnvinorment in UE 4.25.4 but an error message that Main_CityEnvironment.umap appears to be an asset file crops up everytime I try to open this project even after installing the lastest version of airsim and repeadtly downloading the 3 zip files that make up this environment. I can also confirm that the runnable .exe file that comes with the environment works properly and I can control the vehicles normally through API's.

I have followed the official video (https://www.youtube.com/watch?v=1oY8Qu5maQQ) as stated in the documentation, to import the environment. I am aware that this may be an issue with git but I am not sure how to download the environment apart from this link https://github.com/microsoft/AirSim/releases/tag/v1.4.0-windows

The preinstalled Blocks environment works perfectly in UE.

Thank you for any help!

### Settings

{
 "SimMode" : "Car", 
 "SettingsVersion" : 1.2, 
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "Vehicles": {
		"Car1": {
		  "VehicleType": "PhysXCar",
		  "X": 0, "Y": 5, "Z": -2
		}
	     },
	
"DefaultSensors": {
    "Barometer": {
         "SensorType": 1,
         "Enabled" : true
    },
    "Imu": {
         "SensorType": 2,
         "Enabled" : true,
	"PointsPerSecond": 10
    },
    "Gps": {
         "SensorType": 3,
         "Enabled" : true
    },
    "Magnetometer": {
         "SensorType": 4,
         "Enabled" : true
    },
    "Distance": {
         "SensorType": 5,
         "Enabled" : true
    },
    "Lidar2": {
         "SensorType": 6,
         "Enabled" : true,
         "NumberOfChannels": 4,
         "PointsPerSecond": 10000
    }
}

}

### How can the issue be reproduced?

1. Install AirSim 1.4.0 and UE 4.25.4
2. Follow https://www.youtube.com/watch?v=1oY8Qu5maQQ for including the CityEnviron into a UE project

### Include full error message in text form
Failed to load map!
../../../../../../Users/attar/Documents/Unreal Projects/AUTOAI/Content/Main_CityEnvironment.umap appears to be an asset file.

**What's better than filing an issue? Filing a pull request :).**

rajat2004, 2021-04-29:
The released environments are compiled binaries rather than the source assets. AFAIK, there's no way to open the binaries in the Editor and edit them

closed 2021-05-11 by zimmy87

denmonz, 2022-08-25:
@zimmy87 is this an asset we can purchase on the Epic Games Marketplace?