microsoft/AirSim #3657: AirSim with PX4 SITL uncaught exception

gabrielebaris, 2021-05-05:
## Bug report 

- AirSim Version/#commit: commit 0b936af5
- UE/Unity version: UE 4.25
- autopilot version: PX4 v1.11.3
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

I am sorry if this results to be a duplicate, but in any case I tried to fill as many info as possible.
I am trying to run PX4 SITL with AirSim with a binary generated from UE4Editor, but at startup the process crashes with the error:
```
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9
```
This happens both with the Blocks environment and with a custom environment I made.
If instead I start EU4Editor, open the project and run with the play button, that error is caught and the simulation continues.

### Settings

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "Fpv",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlIp": "127.0.0.1",
      "ControlPort": 14590,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14600,
      "LogViewerHostIp": "",
      "QgcHostIp": "",
      "LocalHostIp": "127.0.0.1",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "Barometer": {
          "SensorType": 1,
          "Enabled" : true
        },
        "Imu": {
          "SensorType": 2,
          "Enabled" : true
        },
        "Gps": {
          "SensorType": 3,
          "Enabled" : true
        },
        "Magnetometer": {
          "SensorType": 4,
          "Enabled" : true
        }
      },
      "Cameras": {
        "bottom_cam": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 60,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0
            }
          ],
          "X": 0, "Y": 0, "Z": 0.1,
          "Pitch": -90, "Roll": 0, "Yaw": 0
        },
        "front_cam": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 180
    },
    "ptz": {
    
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_cam", "Visible": true},
    {"WindowID": 1, "ImageType": 2, "CameraName": "front_cam", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "bottom_cam", "Visible": true}
  ]
}
```
I also tried to disable the lockstep mode, but the behaviour is the same,

### How can the issue be reproduced?

1. Run PX4 SITL with `make px4_sitl none_iris`
2. Run AirSim simulation

### Include full error message in text form
```
...
Connected to SITL over TCP.
Connecting to PX4 Control UDP port 14590, local IP 127.0.0.1, remote IP...
[2021.05.05-07.19.10:631][343]LogTemp: Connected to SITL over TCP.
[2021.05.05-07.19.10:631][343]LogTemp: Connecting to PX4 Control UDP port 14590, local IP 127.0.0.1, remote IP...
received first heartbeat
[2021.05.05-07.19.10:669][343]LogTemp: received first heartbeat
Ground control connected over UDP.
[2021.05.05-07.19.10:671][344]LogTemp: Ground control connected over UDP.
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9

Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Failed to find symbol file, expected location:
"/home/gabriele/Documents/Unreal Projects/drone_contest_2y_exp/LinuxNoEditor/LeonardoDroneContest/Binaries/Linux/LeonardoDroneContest.sym"
[2021.05.05-07.19.10:682][344]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2021.05.05-07.19.10:682][344]LogCore: Fatal error!

0x00007f6f1b41718b libc.so.6!gsignal(+0xcb)
0x00007f6f1b3f6859 libc.so.6!abort(+0x12a)
0x000000000853c6fb LeonardoDroneContest!UnknownFunction(0x833c6fa)
0x000000000853c60e LeonardoDroneContest!UnknownFunction(0x833c60d)
0x000000000853c483 LeonardoDroneContest!UnknownFunction(0x833c482)
0x000000000853bd06 LeonardoDroneContest!UnknownFunction(0x833bd05)
0x000000000853bc9f LeonardoDroneContest!__cxa_throw(+0x6e)
0x00000000076f9909 LeonardoDroneContest!_ZN13UdpClientPort13UdpSocketImpl5writeEPKhi(+0xa8)
0x00000000076e4848 LeonardoDroneContest!_ZN15mavlinkcom_impl21MavLinkConnectionImpl11sendMessageERKN10mavlinkcom14MavLinkMessageE(+0x377)
0x00000000076e4f46 LeonardoDroneContest!_ZN15mavlinkcom_impl21MavLinkConnectionImpl11sendMessageERKN10mavlinkcom18MavLinkMessageBaseE(+0x25)
0x00000000076e97e3 LeonardoDroneContest!_ZN15mavlinkcom_impl15MavLinkNodeImpl11sendMessageERN10mavlinkcom18MavLinkMessageBaseE(+0x32)
0x00000000076e8ae7 LeonardoDroneContest!_ZN15mavlinkcom_impl15MavLinkNodeImpl16sendOneHeartbeatEv(+0x56)
0x00000000076e8a48 LeonardoDroneContest!_ZN15mavlinkcom_impl15MavLinkNodeImpl13sendHeartbeatEv(+0x77)
0x00000000076f1978 LeonardoDroneContest!_ZNSt3__114__thread_proxyINS_5tupleIJNS_10unique_ptrINS_15__thread_structENS_14default_deleteIS3_EEEEMN15mavlinkcom_impl15MavLinkNodeImplEFvvEPS8_EEEEEPvSD_(+0x37)
0x00007f6f1be98609 libpthread.so.0!UnknownFunction(0x9608)
0x00007f6f1b4f3293 libc.so.6!clone(+0x42)

[2021.05.05-07.19.10:695][344]LogExit: Executing StaticShutdownAfterError
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Malloc Size=131160 LargeMemoryPoolOffset=459112 
Malloc Size=81112 LargeMemoryPoolOffset=540248 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

jonyMarino, 2021-05-05:
Hi and welcome! @gabrielebaris. Probably you are using a different AirSim version when running with the UE4Editor?

gabrielebaris, 2021-05-06:
Hi @jonyMarino and thanks for the reply.
I can confirm it happens with the same version of AirSim.

jonyMarino, 2021-05-06:
Ok. Can you try disabling your firewall?

gabrielebaris, 2021-05-10:
Hi. Sorry but I couldn't check until now. I think the firewall was already disabled, since `ufw status` is `inactive`. Let me know if you need some more tests

jonyMarino, 2021-05-10:
We have merged many fixes for the PX4 connection. Can you try again, making sure you have the firewall disabled?

gabrielebaris, 2021-05-10:
Sure! Do I have to pull from the latest master?

jonyMarino, 2021-05-10:
Yes!

gabrielebaris, 2021-05-11:
It works!
I created a packaged version of the Unreal simulation using commit 8d21f60e3ac99c4bafa664560f9ea9037228c406 and it works just fine, no more crashes.

closed 2021-05-11 by gabrielebaris