microsoft/AirSim #3663: AirSim with Cesium for Unreal Engine

petergerten, 2021-05-06:
Did anyone else try to use AirSim with [Cesium for Unreal Engine](https://github.com/CesiumGS/cesium-unreal)?

I have it working in principle, but there seem to be some strange effects:
- After takeoff (I use PX4 SITL) at the correct location, Cesium suddenly jumps to a location 1000's of kilometers away. This is reproducible and very strange. My GPS location is just a few meters off the takeoff point.
- When I move my camera off the default direction, the tile renderer does not notice that and won't load new tiles.

petergerten, 2021-05-07:
=> disable 'keep origin near' seems to help

closed 2021-05-07 by petergerten

Rick-Weng, 2021-09-08:
i have the same question of u, which made me sooooooo sad!
May i get the information about where i can find the " keep orign near'?
please~

darkcherryx3, 2024-11-19:
You can find it under the CesiumGeoreference Instance you created.