microsoft/AirSim #3664: Instant Crash of Unity Engine, when following "Custom Unity Environment" Documentation

HackBrettHB, 2021-05-06:
## Bug report 

- AirSim Version/#commit: 0b936af5d76a856f8cc4d5abf9b9f1a61cbeca85
- UE/Unity version: 2019.3.12f1 AND/OR 2020.3.6f1
- autopilot version: N/A
- OS Version: Windows 10 Build 19042.928

### What's the issue you encountered?

If I follow [this guide ](https://microsoft.github.io/AirSim/custom_unity_environments/) from the AirSim Documentation with my custom Unity environment being [this scene](https://assetstore.unity.com/packages/3d/environments/roadways/windridge-city-132222) as suggested [here](https://microsoft.github.io/AirSim/Unity/), Unity crashed instantly when hitting the play button.
The issue persists among the mentioned Unity Versions (The 2019 is out of date, but is suggested in the guide). I have reinstalled and rebooted several times.
The state before I hit play and Unity crashed is this:
[image]

### Settings

None, I freshly installed Unity and AirSim.

### How can the issue be reproduced?

Follow the mentioned guide.

### Include full error message in text form
I don't get an error. Unity completely crashes.
The error araises from AirSimWrapper.dll
I uploaded the full error.txt file from Unity [here](https://pastebin.com/TWnCX05D)

**What's better than filing an issue? Filing a pull request :).**

HackBrettHB, 2021-05-06:
Problem solved! (As always, after many hours of try and error, the problem resolves right after I open an issue...)

The problem is, that the documentation is unclear from my perspective. I tried to execute the CarDemo directly, instead of loading the "SimModeSelector" and executing that.
I would highly recommend to update the documentation [here](https://microsoft.github.io/AirSim/custom_unity_environments/) to be more precise about how to finally execute the code.

I leave the issue open, so you read it. After you have read this, you may close the issue.

rajat2004, 2021-05-06:
Unity doesn't get as much testing around here. It would be best if you could yourself add some details regarding the correct thing to do since you have already tried and fixed this problem, there's an Edit on Github button at the top of the page.

PR #3670 (2021-05-13): Update docs custom_unity_environments.md
Extend description of how to run the project.

Fixes: #3664   

## About
Update documentation to make it clearer for Unity Noobs like me, how to use the project afterwards.

## How Has This Been Tested?
Docs, does not need testing.
closed 2021-05-13 by zimmy87