microsoft/AirSim #3667: AirSim with ROS WSL2 Connection Error: rpc::rpc_error

Ferusel, 2021-05-07:
I've been trying to connect my WSL2 ROS instance to AirSim on Windows.

Following the instructions in the docs: https://microsoft.github.io/AirSim/airsim_ros_pkgs/, when I run the command roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP it gets connected, but starts outputting this error:

```
Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call

[airsim_node-2] process has died [pid 491, exit code -6, cmd /home/ferusel/AirSim/ros/devel/lib/airsim_ros_pkgs/airsim_node __name:=airsim_node __log:=/home/ferusel/.ros/log/e7912b0a-af2d-11eb-b571-00155d953a9d/airsim_node-2.log].
log file: /home/ferusel/.ros/log/e7912b0a-af2d-11eb-b571-00155d953a9d/airsim_node-2*.log]
```

I am currently using WSL2 running Ubuntu 18.04, UnrealEngine4 version  

These are the settings in my settings.json:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "LocalHostIp": "172.17.144.1",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 720,
        "Height": 576,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
    ]
  },
  "Recording": {
    "RecordInterval": 0.01
  }
}
```

If it helps, this was the closest error I think I could find: https://github.com/microsoft/AirSim/issues/3095

rajat2004, 2021-05-07:
The AirSim version isn't mentioned in the issue description, please do update the same!. Assuming you're using the 1.4.0 binaries, and the latest master for the ROS wrapper in WSL, I think the problem is being caused by #3536.
The problem came to my mind only after the PR got merged, the issue is that the latest master ROS wrapper is using an API (`getSettingsString`) which is only in master and not in the 1.4.0 release, and hence the exception.

You could checkout the master branch at commit https://github.com/microsoft/AirSim/commit/6799dc15aef365fd6c2ebb3417d3a4a19c5d3136 (just before the PR merge), and it should work. Note that you'll have to use the exact same settings in Windows as well as WSL in the Documents folder. If it doesn't work then something else is the problem.
I'll try to add some workaround for this, maybe handle the exception and fallback to the previous method? It'll be added in https://github.com/microsoft/AirSim/pull/3635 which I'm using for my ROS testing.

If not using the latest master ROS wrapper, or using source built latest master on Windows also for the simulation, then disregard above, and something else is the issue.

racheraven, 2021-05-22:
We have encountered the same issue today.
We downloaded the Airsim package with `git clone https://github.com/Microsoft/AirSim.git` on a Ubuntu 18.04 computer and a Windows 10 computer, respectively. Therefore, we believe we are currently using the latest1.50 version of Airsim.

 The "AirSim" on Ubuntu 18.04 is the client , and the Win10 pc runs a custom environment on UE4 25.04 with Airsim plugin. 

We got the 

> Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
> terminate called after throwing an instance of 'rpc::rpc_error'
>   what():  rpc::rpc_error during call
> Aborted (core dumped)

on the Ubuntu 18.04 , when we run the `airsim_ros_pkgs` to obtain images from UE environment on Win10.

We followed all the steps on the [Creating and Setting Up Unreal Environment
](urlhttps://microsoft.github.io/AirSim/unreal_custenv/). We can even fly the drone in the scene with PX4 HISL on Win 10.

Btw, if we use another scene made laster year (with an earlier version of Airsim) with the same `setting.json`, there's no error on the client side. And we obtain images successfully on the client side. Therefore, we believe the exception is not caused by networking problems or UE4 errors.

Any idea how we may fix the bug?

Here is the settings file.
```
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "SimpleFlight",
  "LocalHostIp": "192.168.22.30",
  "ViewMode": "",
  "UsageScenario": "",
  "RpcEnabled": true,

  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraID": 0,
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "SubWindows": [
    {
      "WindowID": 0,
      "CameraID": 0,
      "ImageType": 0,
      "Visible": false
    },
    {
      "WindowID": 1,
      "CameraID": 3,
      "ImageType": 0,
      "Visible": false
    },
    {
      "WindowID": 2,
      "CameraID": 0,
      "ImageType": 3,
      "Visible": false
    }
  ],
  "CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 100,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },
    {
      "ImageType": 2,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 100
    },
    {
      "ImageType": 3,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 100
    }
  ],
  "NoiseSettings": [
    {
      "ImageType": 0,
      "Enabled": true,
      "RandContrib": 0.2,
      "RandSpeed": 100000.0,
      "RandSize": 1000.0,
      "RandDensity": 2.0,

      "HorzWaveContrib": 0,
      "HorzWaveStrength": 0.08,
      "HorzWaveVertSize": 1.0,
      "HorzWaveScreenSize": 1.0,

      "HorzNoiseLinesContrib": 0,
      "HorzNoiseLinesDensityY": 0.01,
      "HorzNoiseLinesDensityXY": 0.5,

      "HorzDistortionContrib": 0,
      "HorzDistortionStrength": 0.002
    }
  ],
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": false,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "PX4": {
    "FirmwareName": "PX4",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135,
    "ApiServerPort": 41451
  }
}

```

rajat2004, 2021-05-24:
I think the problem lies with the combination settings being used in both the cases, and with a behaviour change in AirSim parsing them, see https://github.com/microsoft/AirSim/pull/3726
@Ferusel The `SimMode` isn't present in the settings, without that, the ROS wrapper defaults to the Car mode. Adding a vehicle element should fix the error, assuming that it is being caused due to the sensor APIs and not something else.

@racheraven Unless you're using a modified version of AirSim, many of the fields in your settings don't exist at all. The PX4, SimpleFlight elements should be inside `Vehicles`, `FirmwareName` should be `VehicleType`, etc. Adding the vehicles inside the `Vehicles` section should fix the error

imanf94, 2021-07-08:
I have a similar problem. I am working on connecting ROS (in WSL2) to AirSim on windows 10. My AirSim environment is the City latest release (v1.5). The ROS on ubuntu 18 is also working properly. When I try to run this code `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP` the connection established successfully but I get this error which led the process to die.

roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP
... logging to /home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/roslaunch-DESKTOP-4BJ7Q09-377.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://DESKTOP-4BJ7Q09:40949/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: 172.30.64.1
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[airsim_node-1]: started with pid [392]
process[ned_to_enu_pub-2]: started with pid [393]
Waiting for connection - X
**Connected!**

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
  what():  rpc::rpc_error during call
[airsim_node-1] process has died [pid 392, exit code -6, cmd /home/iman/AirSim/ros/devel/lib/airsim_ros_pkgs/airsim_node __name:=airsim_node __log:=/home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/airsim_node-1.log].
log file: /home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/airsim_node-1*.log

And here is my settings.json files which are located in Documents/AirSim folder of both Windows 10 and WSL/Ubuntu:
```

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "PhysicsEngineName": "FastPhysicsEngine",
  "ClockType": "SteppableClock",
  "ClockSpeed": 0.34,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": true,
      "EnableCollisions": false,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "LidarSensor": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "Range": 30,
          "RotationsPerSecond": 10,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "Roll": 0,
          "Pitch": 0,
          "Yaw": 0,
          "VerticalFOVUpper": 5,
          "VerticalFOVLower": -5,
          "HorizontalFOVStart": 180,
          "HorizontalFOVEnd": -180,
          "DrawDebugPoints": false,
          "DataFrame": "SensorLocalFrame"
        },
	"Imu" : {
          "SensorType": 2,
          "Enabled": true
        }
      },
     "Cameras": {
        "leftcamera_1": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": -0.06, "Z": -2.4,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "rightcamera_1": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma":  @1.5
            }
          ],
          "X": 0, "Y": 0.06, "Z": -2.4,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 1, "ImageType": 0, "CameraName": "leftcamera_1", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "rightcamera_1", "Visible": false}
  ]
}

```
I guess my settings.json has what @rajat2004 mentioned. Anybody knows how can I fix this?