microsoft/AirSim #3679: The depth camera doesn

Deepakgthomas, 2021-05-10:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:
- autopilot version:
- OS Version:

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

Deepakgthomas, 2021-05-12:
@jonyMarino Can you please help with this issue?

jonyMarino, 2021-05-12:
Hi @Deepakgthomas! It may be a non-trivial fix. Why do you need it for?

Deepakgthomas, 2021-05-12:
Thanks for replying Jonathan! The depth images are used as input to my RL
algorithm. I wanted to create some kind of noise in my environment such as
snow, rain, leaf etc. Basically change the condition/ weather of my
environment.

zimmy87, 2021-05-14:
Hi @Deepakgthomas, we experimented with modifying all the collision parameters inside the particle system for snow: ParticleSystem'/AirSim/Weather/WeatherFX/Particles/P_Weather_SnowFX.P_Weather_SnowFX'. Unfortunately, we couldn't figure out a configuration that would make the particles visible to the DepthViz or Segmentation post process materials. We believe that the DepthViz depends on getting the position of the individual particles which is not currently accessible in the WeatherFX. It seems like it would be easier to generate noise for these visualizations in a different way, and you are welcome to submit a feature request.

closed 2021-05-14 by jonyMarino